EdgeTX/edgetx #648: TS16 firmare installing with hardware ADC on... (bug)

sugaarK, 2021-08-28:
so helping a friend who has elrs on his ts16 running 250hz and he has a lot of ff jitter in this flights.. I straight away recognised  the issue looking at the log and the set point stepping as the ADC been on in the Hardware settings of his Ts16..
you guys should be disabling it for any radios that have hall gimbals or direct people to turn it off..

here flight with it on note the unstable FF
[image]

and here is a bench test with it off 

[image]

thanks guys!!

sugarK

raphaelcoeffic, 2021-08-28:
@sugaarK what should we disable by default exactly? Do you mean the ADC filter?

sugaarK, 2021-08-28:
> @sugaarK what should we disable by default exactly? Do you mean the ADC filter?

I'm not sure if the same firmware build is used for the higher spec radios and the basic model.. but yeah for 99% of people you dont want the ADC on it causes setpoint stepping and this drives the motors up and down via the FF code. as I said if its more practical to leave it on ( for pot based gimbals ) then letting people know to turn it off for Hall effect gimbals would be clutch...

raphaelcoeffic, 2021-08-28:
> I'm not sure if the same firmware build is used for the higher spec radios and the basic model.. but yeah for 99% of people you dont want the ADC on it causes setpoint stepping and this drives the [...]

ADC means "Analog Digital Converter", and this is used anyway, otherwise you have no gimbal values at all. Now the option you are probably talking about is placing a filtering algorithm on top on the value collected from the ADC. Is this what you are talking about? Please note that hall-gimbals also use analog values (with very few exceptions, where the analog-digital conversion is done in the gimbal itself).

raphaelcoeffic, 2021-08-28:
Ok, after discussing this with @sugaarK it seems this is really about having the ADC filter turned OFF by default.

rotorman, 2021-08-28:
So basically jitterFilter
https://github.com/EdgeTX/edgetx/blob/8becbfcb499d919b6ba537de06778ab296a734e8/radio/src/datastructs.h#L839
being 1 (active) for hall gimbals and 0 (disabled) for pot gimbals?
The problem is, there is no automagic way to detect if the physical radio, e.g. RM TX16S, has pot or stock hall gimbals.
As only very few TX16S were sold with pot gimbals, one could assume here hall gimbals though.

But for my understanding -> why would jitter suppression then be necessary on typically more stable pot gimbals at all?

gagarinlg, 2023-03-18:
Is this still an issue? We can now enable/disable ADC filters per model, which makes a lot of sense for people have things with and w/o FCs

pfeerick, 2023-03-19:
I don't think so. As you say, it can be done selectively per model, and also via model templates - i.e. there is nothing stopping anyone from contributing templates to be distributed with the EdgeTX card pack specifically for multi-rotor (or added after the fact - i.e. a "stock" Betaflight config ;) ).

closed 2023-03-19 by pfeerick