EdgeTX/edgetx #650: widget colour

pds59, 2021-08-28:
radiomaster t16s edge tx version 8751004 can not change colour of the widget.  some widget are gray, some white, some black, some blue, quite difficult to read

pds59, 2021-08-28:
[image]
[image]
i enclose picture

rotorman, 2021-08-28:
Which EdgeTX version you are using (GitHub hash)?

pds59, 2021-08-28:
Firmare ; main

Version: //e216921(2021-08-28t10:54:33z)

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

pds59, 2021-08-28:
on radio system i read: 
vers: 2.5.0 (e2169215)

pfeerick, 2021-08-28:
#646 should fix the ability to change lua widgets, and #649 (once merged) will make it so lua widgets check the colour when created.

pfeerick, 2021-08-29:
#649 has just been merged, so please try with the current main, or the next nightly. It only effects the loading of the custom colour used by Lua Widgets (i.e. Timer2, Counter, BattCheck, TxGPStest, etc.). 

For other widgets, such as Text and Value, they are firmware provided widgets, and their colour settings (in the individual widget options) are working for me on TX16S with e216921. They can be set, and I often find the 'Shadowed' option make a big difference to readability.

pds59, 2021-08-29:
ok will try and let you know

pds59, 2021-08-30:
solved tks

closed 2021-08-30 by pds59