EdgeTX/edgetx #658: edit one mix remove all the next mixes in the list

ishay320, 2021-08-30:
**Describe the bug**
the bug is when edit the source of a mix to be nothing (-), the list from that element is removed.

**To Reproduce**
open model menu and go to mixes,
select mix from the middle of the list and edit it,
in source select nothing (-)
return

**Expected behavior**
it should keep the tag and all other mixes.

**Screenshots**
the original list:
[image]

edit it to be nothig:
[image]

the list after:
[image]

**General information**
 - EdgeTX version2.4.0
 - Hardware RM TX16S

eshifri, 2021-08-30:
Is it the same if you just remove this mix?

pfeerick, 2021-08-30:
On basically yesterday's nightly, if you delete a mix line, other rows are unaffected. This bug is also reproducible there, but not on B&W - perhaps partly because on B&W once you have set "a" source, it can't ever be set to "---" again.

eshifri, 2021-08-31:
B/W does not use libopenui, so it is completely different implementation of the mixies screen.

pfeerick, 2021-08-31:
For the UI, yes, but this does not automatically rule out an issue with shared/common functions.

eshifri, 2021-08-31:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/model_mixes.cpp#:~:text=for%20(uint8_t%20ch,grid.getFieldSlot()%2C%20mixIndex)%3B

  for (uint8_t ch = 0; ch < MAX_OUTPUT_CHANNELS; ch++) {
    if (mixIndex < MAX_MIXERS && mix->srcRaw > 0 && mix->destCh == ch) {

      coord_t h = grid.getWindowHeight();
      auto txt = new MixLineTitle(window, grid.getLabelSlot(),
                                  getSourceString(MIXSRC_CH1 + ch),
                                  BUTTON_BACKGROUND, COLOR_THEME_PRIMARY1 | CENTERED);

      uint8_t count = 0;
      while (mixIndex < MAX_MIXERS && mix->srcRaw > 0 && mix->destCh == ch) {
        Button * button = new MixLineButton(window, grid.getFieldSlot(), mixIndex);

pfeerick, 2021-08-31:
I'm not so sure it is this... yet as it's not cosmetic - the mixes are actually gone. If you check the channel monitor there is no output anymore :-/. i.e. if you have four channels, edit the second, set the source to `---` and then exit the page... you only have one active channel.

eshifri, 2021-08-31:
If I fix this code, mixes are shown and preserved, but all mixes after the one without source become inactive and channel monitor does not show them.

pfeerick, 2021-08-31:
Yes, that is what I feel is the real problem... I just haven't been able to place where exactly the mixes are being deleted/inhibited.

eshifri, 2021-09-01:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/mixer.cpp#:~:text=for%20(uint8_t%20i%3D0%3B%20i%3CMAX_MIXERS,break%3B

If this break is changed to continue, everything works.
I do not know what protection mechanisms this can break. :-)

pfeerick, 2021-09-01:
You mean protection mechanisms can break? 😆 
https://github.com/EdgeTX/edgetx/blob/cea723c56ea52db45911e4e5cbe4485cb40c23b4/radio/src/mixer.cpp#L650-L651

At least you found the culprit, now it's just a matter of determining the ~~punishment~~ fix for it. :) 

btw, for sharing links to sections of code, it is much better if you if you can use the line number / permalink option in github... it then shares the line number in the URL, and even interpets it itself in issues like above. You can click on the line number for a single line, or click on the first line, and then shift+click to highlight a code range.

[image]

PR #665 (2021-09-03): allow mixes without source
This resolves #658 allowing mix with "--" source in the middle of the mixes list.
The b/w version is not affected. :-)
closed 2021-09-03 by pfeerick