EdgeTX/edgetx #663: widget colors display problem

jogybaer0815, 2021-09-01:
Radiomaster TX16S
Vers. Nightly 3533fd1a / 2021-08-31

Problem widget: Text-Colors are not displayed inverted under the% bars - they appear overwritten.
Regards

In the picture channels 04 and 05:
[image]

pfeerick, 2021-09-01:
Thanks for this. :) I will be reviewing this soon to see if something similar to the Guages widget can be done here - where it inverts the text colour as the bar overlaps the text. This was always an issue with this widget, but more noticable now since the bar and text are the same colour.

JimB40, 2021-09-02:
@pfeerick Invert part of text which is masked to preserve UI space for text. We talked abt that :)

pfeerick, 2021-09-02:
Tried to make that work, and it's not behaving for me atm, so this is next best for now. Problem for when have time to figure it out properly.

MervDale, 2021-09-04:
I know it was mentioned to me that the OTX colour coding left a bit to be desired but seeing as it works and can give a fairly big range of colour options would it not be better trying to adopt and tidyup the OTX colour system, unless of course ETX coding has veered to far away. Rather to try and re invent a working system?

pfeerick, 2021-09-04:
ColourLCD support 2.3 => 2.4 (which is what ETX is based on) is a work in progress complete re-implementation of the user interface using a common UI library rather than custom code all the way through. The colour backend has now basically been fixed and is ready for theming. Now the widgets need fixing - i.e. outputs widget was *never* good at the best of times... it doesn't do "basic" things like invert the text colour when the bar goes over it, so either the default colours had to be good (thus limiting colour choices) or you needed to be able to choose the colours yourself (which wasn't made available).

PR #664 (2021-09-09): Set more sensible default colors for output widget
Also allows text and bar colors to be customised!
Fixes #663
commit 49b91431 (2021-09-09): Set more sensible default colors for output widget
And also allow colors to be customised!
Fixes #663
closed 2021-09-09 by pfeerick