EdgeTX/edgetx #667: [LUA] lua.drawTextFormatted

JimB40, 2021-09-02:
1. This will be very useful for long text, description errors
2. We can improve it later to read markup lang but for now simple split at width character number will work

So from my LUA impemetation
1. Define additional params (max_characters, line_space)
2. If line character_count > max_characters I'm searching fo first space before max_character and split there if not found (logs, errors etc) split at max_character
3. line_space optional to set look and feel

pfeerick, 2021-09-02:
That would be https://github.com/jfrickmann/edgetx/blob/71764fec1d8f1962c2d324965d88c46110568d96/radio/src/gui/colorlcd/draw_functions.cpp#L507-L530

as part of #653, but it would be nice to see a smarter Lua draw method for longer text blocks. 

I suppose it needs to know the X,Y,W,H + text to display? So it displays as much of the text as fits the W, and wraps it where needed?

JimB40, 2021-09-02:
Yep but due to color radios flex font width it may be simpler to start with max_character limit

jfrickmann, 2021-09-02:
The function that Peter refers to could easily be used by a new Lua API function as well:
```lcd.drawTextLines(x, y, w, h, text, flags)```
It draws the lines of text inside a bounded rectangle, and will just stop if there is not room for all of the text. It gets the line distance from the flags, so you don't have to worry about that.

JimB40, 2021-09-02:
Great. Don't read code every day. :)
@jfrickmann what you say to change fn name to lcd.drawTextBox?
Once we start to use it in LUA it won't be changed any more :)

jfrickmann, 2021-09-02:
I added it to PR #671. I kept the name because it matches the C++ function and it does not draw a box.

pfeerick, 2021-09-15:
`drawTextLines` exists now, so closing

closed 2021-09-15 by pfeerick