EdgeTX/edgetx #669: Problem with colors in Switch2 widget

stefaanbolle, 2021-09-02:
Using the latest commit, when using the switch2 widget [https://www.rcgroups.com/forums/showpost.php?p=47009445&postcount=3793](url) the widget settings does not allow proper color selection.
With the latest commit, the virtual keyboard when editing these RGB values hangs.
R value max is 31 while G value max is 63 on my TX16S.
Impossible to get OFF color to red (RGB 255/0/0)

stefaanbolle, 2021-09-02:
just realizing this might be a similar same issue as #666

pfeerick, 2021-09-02:
I think it is... I can change the colours, but the keyboard hangs. Fix is pending though ;). In the mean time, you can change your colours, and power cycle the radio *before* it Emergency Modes or crashes ;)

Duplicate of #666.

closed 2021-09-02 by pfeerick

pfeerick, 2021-09-02:
Handy looking widget btw... thanks for pointing it out! :)