EdgeTX/edgetx #670: Bind procedure doesn't work on T18 for R9M ACCESS

ViorelX, 2021-09-02:
I have compiled EdgeTX for T18 target with 'externalaccessmod' feature.
I tried to get working a R9M module with a R9mm receiver. I can do the 'register' procedure, it works and finally appears the 'registration OK' message, but the binding process cannot be accomplished.
I have to revert to OpenTX to do this, then I install back EdgeTX and the receiver remain bound.

ViorelX, 2021-09-03:
Moreover, I have noticed:
-there is not the 'options set' button for the R9M transmitter;
-the 'receiver no' field is also missing;
-the transmitter does not transmit output channel data to the receiver (but it transmits telemetry).

gagarinlg, 2023-02-26:
fixed with #991

closed 2023-02-26 by gagarinlg