EdgeTX/edgetx #673: Numeric keyboard does not have erase button.

erwinf, 2021-09-03:
I just updated to the latest nightly and noticed that when switching to numeric keyboard, there is no backspace option. I had to switch to the abc keyboard to delete incorrectly entered numbers.

Radio: TX16S

kevinkoenig, 2021-09-03:
That was the original implementation.  It was not modified as part of adding keyboard support.  Should we implement a clear button(), or a backspace.  The issue is like for timers which are numbers with a specific display, backspace might not make sense.

From: erwinf
Sent: Friday, September 3, 2021 10:47 AM
To: EdgeTX/edgetx
Cc: Subscribed
Subject: [EdgeTX/edgetx] Numeric keyboard does not have erase button. (#673)

I just updated to the latest nightly and noticed that when switching to numeric keyboard, there is no backspace option. I had to switch to the abc keyboard to delete incorrectly entered numbers.
Radio: TX16S

eshifri, 2021-09-03:
I think this is about the keyboard per se - not number_edit.
Actually, there is a backspace there - it is just left at the same place where it was for abc keys.

pfeerick, 2021-09-04:
NumericEdit, so it's the number keyboard. ;)

I've noticed the DEF/default button seems to go to zero a lot? Maybe that
is sufficient?

pfeerick, 2021-09-04:
NumericEdit / "when switching to numeric keyboard" ;)

I've noticed the DEF/default button seems to go to zero a lot? Maybe that
is sufficient?

[image]

eshifri, 2021-09-04:
I think it was about full keyboard. The first screen has letters; the second - digits. You cannot delete "22" by switching to "abc" keyboard - no such option. :-)
I think the complaint is e.g. you are editing model name: Edge540. To enter 540 you have to go to the second screen (touch digits button). Delete button is there, but it is at the very bottom - not easy to notice.

rotorman, 2021-09-04:
@kevinkoenig I just post the info here - we have following remark from a fellow Discord user about the keyboard caps key:
`Hi all, just loaded nightly t18-69efe13 and the keyboard is great my only question so far is the caps arrow is solid in colour when in lower case and not coloured when in upper case which to my way of thinking is reversed as compared to my android mobile phone, not a complaint just an observation:-)`
https://discord.com/channels/839849772864503828/839849772864503832/883579134570266654
Would be sweet if you would have the joy of looking into it. Cheers!

MervDale, 2021-09-05:
Hmm maybe they should look closer and then ask the question elsewhere first before raising it on GitHub as an issue, just a thought ;-Q

pfeerick, 2021-09-05:
@eshifri My apologies... I read that too early in the morning before heading out flying... 😆 

Backspace is indeed still there... so 👓 needed 😆 

[image]

closed 2021-09-05 by pfeerick