EdgeTX/edgetx #674: Nightly is not working on Jumper T16

Salto42, 2021-09-03:
Nightly is not working on Jumper T16 since 31/08/21.

pfeerick, 2021-09-04:
What are the symptoms? Doesn't start at all? Gets to splash screen and freezes? Welcome audio plays? Emergency mode?

Salto42, 2021-09-04:
The radio does not start: Led1 on, screen black.
You have to disconnect the battery to restart with the previous nightly.

pfeerick, 2021-09-05:
Ouch. This is the first report of that. 

So to eliminate other variables... how are you loading EdgeTX onto your radio? Bootloader flashing or via Flasher? If you're using flasher, is it the latest version of Flasher? As older versions had a problem with flashing the wrong firmware entirely. The last changes to Flasher were nine days ago - and changed the radio selection bar to a dropdown, as well as fixing the issue where it kept reverting to flashing TX16S firmware whilst still highlighting whatever option you selected. I also made a change that is probably the same time ast this, adding manufacturer names to the dropdown, and there is a possibility those changes are not compatible with older versions of Flasher using the horizontal scrolling selector. 

This is the current version for reference:
[image]

Salto42, 2021-09-06:
I have used both solutions to load the software and with the latest Flasher and I always get the same result.
On the other hand I can follow the evolution of EdgeTX with my Radiomaster TX16S without any problem.

tourerjim, 2021-09-06:
I've done edgetx main and edge no issues, only upload using the pros video, copy .bin to sd card content, open firmware file, paste, unplug usb, then in sd card, firmware, .in file, flash, turn tx off and turn on in boot load, click on .bin file and flash, job done.

MRC3742, 2021-09-08:
@pfeerick 
Tested nightly on Jumper T16 not working since 31/08/21 as described by Salto42 above.

t16-4ad5cc0.bin is the last working commit for the T16
All test builds and nightlies using main branch starting with t16-cb26ed3.bin will not allow the T16 to start.
Must disconnect battery, start in bootloader mode, reflash t16-4ad5cc0.bin or an earlier version.

Possible conflict with ? 
https://github.com/EdgeTX/edgetx/blob/24f88092b31704199e8cb4aae2f8ea58cd232477/radio/src/targets/horus/CMakeLists.txt#L160-L161

pfeerick, 2021-09-08:
Entirely plausable, since `SOFTWARE_KEYBOARD` was added in basically the next PR/commit. It does mean that for some reason `HARDWARE_TOUCH` is turned on for hte T16 builds, which it shouldn't be. 

Would you mind trying this build? It's basically the nightly, and I've simply added `-DHARDWARE_TOUCH=OFF` to build options used by the github action... 

[firmware_HARDWARE_TOUCH_OFF.zip](https://github.com/EdgeTX/edgetx/files/7125689/firmware_HARDWARE_TOUCH_OFF.zip)

pfeerick, 2021-09-08:
This is a zipped version with both `HARDWARE_TOUCH` and `SOFTWARE_KEYBOARD` OFF for completeness... I don't think there is any real difference as touch *should* have disabled the keyboard... 
[firmware_HT-OFF_SK-OFF.zip](https://github.com/EdgeTX/edgetx/files/7125729/firmware_HT-OFF_SK-OFF.zip)

MRC3742, 2021-09-08:
Both of these files still will not start on the T16, The same as problem versions.
A short flash of the screen then nothing but black and the #1 led of the six button stays on.
Must disconnect battery, start in bootloader mode, reflash t16-4ad5cc0.bin or an earlier version.

Thanks - Rich

pfeerick, 2021-09-08:
Round 2 ;) I managed to get CMake to admit it wasn't doing what I told it... so can fix that properly now if this does indeed work... it looks promising as the binary is smaller now... 

[firmware.zip](https://github.com/EdgeTX/edgetx/files/7125974/firmware.zip)

MRC3742, 2021-09-08:
Yes - That one now works!!! You found the problem.
Shows Version 2.5.0 (24F88092)

No trauma! :)  
Thanks -Rich

pfeerick, 2021-09-08:
Fantastic! Assuming it didn't break something else for another transmitter build... the nightly after this is merged will work also then. :) https://github.com/EdgeTX/edgetx/pull/690

closed 2021-09-08 by pfeerick