EdgeTX/edgetx #675: Emergency mode on NV14

eshifri, 2021-09-03:
**Describe the bug**
I have a feeling :-), that NV14 never switches into EMERGENCY MODE.
For all cases when TX16S does it, NV14 just crashes.
Then it boots into emergency mode.
It was very clear with the problem of widget color editing with virtual keyboard for number_edit.

**General information**
 - EdgeTX version current
 - NV14

pfeerick, 2021-09-05:
I so want to mark this as a feature... "See, this radio doesn't Emergency Mode!" 😆

pfeerick, 2021-09-20:
Extra bit of info on this... it appears (in some cases, I'm not sure if it's in *all* cases), that EM will be shown if the NV14 is powered via USB... in other words, it resets properly as the power doesn't get cut... found this out when testing the AFHDS3 hiss, crackle and crash bug...

eshifri, 2021-09-20:
Can it be related to this difference?
https://github.com/EdgeTX/edgetx/blob/main/radio/src/targets/common/arm/stm32/pwr_driver.cpp#L135

https://github.com/EdgeTX/edgetx/blob/main/radio/src/targets/nv14/pwr_driver.cpp#L88

raphaelcoeffic, 2021-10-04:
The problem is basically here:
https://github.com/EdgeTX/edgetx/blob/fc2142ec78b81b763c8143395cf1a3ddf14271b0/radio/src/targets/nv14/pwr_driver.cpp#L89

The RCC register holds the reset reason, so that, if cleared on boot, the emergency mode can never be triggered.

closed 2021-11-01 by raphaelcoeffic