EdgeTX/edgetx #683: Colourlcd - Model Outputs - GV for Subtrim, Min, Max values

pfeerick, 2021-09-06:
OTX allows for a Models channel Outputs subtrim, Min and Max values to be linked to a GV... i.e. 
[image]

Whereas this does not currently seem possible on the colourlcds at pressent... (yes, I know this is NV14 screenshot, but same applies) ... perhaps a similar long press (and long touch?) toggle? 
[image]

eshifri, 2021-09-06:
Can't we do it just with class GVarNumberEdit?
Similar to mixes.

eshifri, 2021-09-06:
I'm submitting PR; please take a look.

closed 2021-09-12 by pfeerick