EdgeTX/edgetx #694: Compile Companion for Macos

stefaanbolle, 2021-09-09:
I currently compile TX16S ETX firmware using the Ubuntu20.04 setup as described in the wiki. This works great.
Can someone specify how I can compile Companion for Macos using Ubuntu20.04 or is this not possible?

pfeerick, 2021-09-09:
Cross-compile - I don't know if that is possible. 

You may be able to glean some idea as to what's needed to build on MacOS itself though from the mac os github workflow in this PR... https://github.com/EdgeTX/edgetx/pull/561/files#diff-f8c72078837c2b083745e8f21c7cea9016d9e7ed9cbcf6e57e22c93ce7e979fd

raphaelcoeffic, 2021-09-09:
> I currently compile TX16S ETX firmware using the Ubuntu20.04 setup as described in the wiki. This works great.
> Can someone specify how I can compile Companion for Macos using Ubuntu20.04 or is this not possible?

It’s probably better to compile natively. It’s quite easy in fact using “brew” to install the dependencies.

stefaanbolle, 2021-09-09:
> It’s probably better to compile natively. It’s quite easy in fact using “brew” to install the dependencies.

Can you give some more how-to details?

raphaelcoeffic, 2021-09-12:
> Can you give some more how-to details?

In case this is interesting for you, we have this: #561. It does not use `brew`, but produces a valid companion `.dmg`. You can basically install the same dependencies with `brew`.

raphaelcoeffic, 2021-09-13:
@stefaanbolle I added this here, please let me know if that works for you: https://github.com/EdgeTX/edgetx/wiki/Build-instructions-using-macOS-10.15-%28Catalina%29

stefaanbolle, 2021-09-14:
Excellent, thanks!

closed 2021-09-14 by stefaanbolle