EdgeTX/edgetx #697: Repeatable Keyboard Crash

MervDale, 2021-09-10:
Unfortunately I have found that the keyboard can be crashed on the TX16S in that it requires a power down to get out of the crash.
Top bar widget set to “Text” while editing the text if I didn’t just use the backspace option but instead moved the cursor with my finger and touched the keyboard at the same time while placing the cursor over my edit character I could get the keyboard to drop the top keyboard line and crash,  this is repeatable !

[image]
[image]

*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-09-10:
@kevinkoenig You broke it... how does the rest of the line go? 😆 

Seems to be related to widget options... I can also reproduce this on TX16s... seems to be if you hit the keyboard at basically the same time as the edit field it goes bonkers (yes, that is the technical term for it). 

I can go one step further... and get an interesting half-height screen effect... only seems to effect the main screen when it happens - i.e. radio setup and model setup screens go full screen again. 

https://user-images.githubusercontent.com/5500713/132810898-31b97b7e-eec0-4444-bc4b-8b552b53eb1e.mp4

MervDale, 2021-09-10:
You can really screw thing up trying to clear it as you can still bring up the menu and change models after you send it "Bonkers"
[image]

kevinkoenig, 2021-09-10:
Let me look at it today

kevinkoenig, 2021-09-10:
I just pushed a PR to libopenui #19 that protects against the craziness that was reported and verified by @pfeerick.  Interesting defect.  Not totally my fault :), as the way the keyboard was originally written, it finds the first full screen window.  It does this just by testing the height and width of the window.  

When the keyboard first comes up, it resizes that window.  In onTouchEnd it redisplays the keyboard, like it should, and it finds a different full screen window, and happily resizes that one too, until it finds the main window.  Once that happens all hell breaks loose.  

The PR in libopenui #19 protects against that by testing if the keyboard is already displayed.  If so, it does nothing!  It is interesting to note that in all other screens I could find on every onTouchEnd event it hides the keyboard first and then redisplays it, which is why it doesn't fail anywhere else.  

Fun stuff.  The PR for libopenui should get pushed quickly. 

I also added support to hide the keyboard (like every other window) in topbar.cpp onTouchEnd.  This makes the code more consistent.  Since it is less critical i just put that in the existing branch that there is an open pr against. Once that gets merged it will clean up the keyboard support.

MervDale, 2021-09-14:
Have you guys managed to fix this crash issue with the latest nightly ? Try as I might i could not get it to crash or repeat any of my above screens

kevinkoenig, 2021-09-14:
The fix for this issue was merged last night.  You should not be able to reproduce it any more

rotorman, 2021-09-14:
@MervDale If that is the case for you, can we close this issue?

pfeerick, 2021-09-14:
My mistake, I didn't link to this to the choicex pr as it was being fixed as part of that. This is indeed fixed.

pfeerick, 2021-09-14:
Fixed by #691

closed 2021-09-14 by pfeerick

kevinkoenig, 2021-09-18:
I have been looking at this for the past hour.  I know what is happening.  Now just to fix it.  

It turns out the duplication case is really easy.  Just click 4 times in the field on Widget settings.  On the 5th click if you have the mouse x coordinate over the enter key (even though the y is over the text field) it fails.  

Working on it.

Kevin

From: Kevin
Sent: Friday, September 10, 2021 4:19 AM
To: EdgeTX/edgetx
Cc: EdgeTX/edgetx; Kevin Koenig; Mention
Subject: Re: [EdgeTX/edgetx] Repeatable Keyboard Crash (#697)

Let me look at it today

kevinkoenig, 2021-09-18:
Let me look at it today.
Well that was a fun morning.  I just pushed a PR to libopenui that protects against the crazyness that was reported and verified by Peter.  Interesting defect.  
Not totally my fault 😊
The way the keyboard was originally written, it finds the first full screen window.  It does this just by testing the height and width of the window.  Well when the keyboard first comes up, it resizes that window.  In onTouchEnd it redisplays the keyboard, like it should, and it finds a different full screen window, and happily resizes that one too, until it finds the main window.  Once it finds the MainWindow all hell breaks loose.   
I pushed a PR to libopenui that protects against this case.