EdgeTX/edgetx #698: Category Editor

dlktdr, 2021-09-10:
I've been working on a Category Editor https://github.com/dlktdr/edgetx/tree/category_editor and so far have the ability to Add / Remove / Rename and Move models between categories working. Onc glitch if you add one and instantly rename it will crash I still have to figure out yet.
**but.**.. real reason for making this issue is I'm looking for advice on proper method on what to do about Icons (Even though probably don't need really need them ;)

1) What kind of icon to use.
   - Completely customizable image (.png for instance) 
   - Index into some pre-defined ones such as how 480_default loads the menu icons, just add a few new ones (should support any theme this way too. The method I would lean towards. Can just make a scrolling widget so user can pick one. ( Plane / Drone / Car / Truck, etc.. )

2) How to save the information
I assuming would have to be saved into RADIO/models.txt. Was thinking directly below the header if there is a icon= such as below it will call it up, if nothing use the default Plane. Kind of like an ini file, maybe a : to signify its not a file. But I'm not sure if this is going to cause issues with how companion or future ETXCompanion is going to work?

[Drones]
:icon=23
model1.bin
model2.bin
[Planes]
model3.bin
model4.bin

Anyway just looking for some advice as I learn how this all works!

[image]
[image]

wimalopaan, 2021-09-10:
Looks reasonable to me. Another idea would be to switch to yaml-format here also?

pfeerick, 2021-09-10:
As wimalopaan, YaML would be the way to go for this, although it does obviously mean this would have to wait for YaML to be integrated, so would be post 2.5... which may not be a bad thing to be honest, as we really should be winding up for RCs now and playing wack-a-bug...

dlktdr, 2021-09-10:
Is YAML integrated in main

JimB40, 2021-09-10:
We scratched categories in favor of labels. So this should go this way. @raphaelcoeffic ?

pfeerick, 2021-09-11:
Ah, yes, thanks for the reminder. Well in that case... how about the following:
- go with this as it is now (minus the icons) - thus allowing naming of category, moving of entries between categories and deletion of categories with any mucking around with manual editing of `models.txt`
- build upon this to do labels instead, once YaML is in effect, thus adding label and graphic support, and any UI changes that this brings

pfeerick, 2021-09-13:
@dlktdr Is this in state where we could use it basically now, and not worry about the icons and change to the `.ini`, since idealy we'd change over to using labels once `.yml` is in play (and will also have more flexibility about settings then).

dlktdr, 2021-09-14:
As long as we can get the issue fixed on #714 it was working for me. I haven't had time to look into the reason for that issue yet.

There was a small issue you can't add a new model to a blank page by touching only hardware press, but I think that's a quick fix to pop up the menu if there are no models yet.

pfeerick, 2021-09-14:
Fantastic... Kevin is traveling atm but seems the know the keyboard stuff well, so after digging around there for a while, so it should hopefully be a easy fix -i.e. perhaps simply firing the same event as the hardware enter key enter... after all, I don't there there are any multi-line inputs needing a newline?

kevinkoenig, 2021-09-14:
I already looked at it quickly.  That is what is needed.  What I really need is a way to test the change, as I don’t see anyone using that event right now.  Can someone give me a branch to pull that I can use to test? And the test case also please

dlktdr, 2021-09-14:
Very simple test here. in model_setup.cpp

[image]

The pop-up won't happen on the soft key enter right after you change the models name, only after you touch elsewhere or use the hardware buttons.

If done with hardware keys and rtn pressed pops up right away.
Or if text is changed with the keyboard and press rtn also works.

Thanks @kevinkoenig

closed 2021-09-15 by dlktdr