EdgeTX/edgetx #700: [jumper t-lite] Betaflight Lua scripts

nightflasher, 2021-09-10:
Hey there,
The jumper t-lite shows only dashes in the betaflight Lua scripts (v1. 5.0). The openTX version 2.3.14 worked fine with the betaflight Lua scripts (I was told).

pfeerick, 2021-09-17:
I'm running a build from about three weeks ago on my T-Lite, and betaflight lua 1.5.0 is working for my 4.2.9 whoop using ELRS using 1:16 telemetry ratio. The dashes appear for a few seconds, and then go away, and I have "no telemetry" flashing at the bottom, but it works - reading current values for VTX and saving them. I could probably make it work better by giving more bandwidth to the telemetry but it's an acceptable tradeoff.

gagarinlg, 2023-03-18:
Can this be closed?

pfeerick, 2023-03-24:
I think so yes, since there was no further comment. Additionally, I've just tried on T-Lite with ELRS 2.8.1, ELRS 3.2.0, BF 4.4.0, BF Lua 1.7.0 and seems fine.

closed 2023-03-24 by pfeerick