EdgeTX/edgetx #701: [NV14] Bluetooth - joystick for simulators

nightflasher, 2021-09-10:
There are at least two different Bluetooth modules build into the flysky nirvana, somebody managed to get the darker module working with Android and the telemetry viewer app. However, if like to use this app with my blue Bluetooth module and maybe it's possible to provide a wireless joystick over Bluetooth for mobile simulators like freerider or CurryKitten flightsim.

The nirvana is a good radio for flying, but I don't want to deal with a long cable while getting some stick time in the Simulator. Also, I don't want to buy long cables anymore or carry them with me.

yds, 2021-09-11:
[Flysky-OpenTX-Test-Branch NV14_Bluetooth commits](https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/commits/nv14_bluetooth) branch relevant to this issue

darshanmahakul, 2024-01-31:
anyone know how to use bluetooth joystick with edgetx setup

pfeerick, 2024-02-01:
Currently not possible... bluetooth module isn't enabled by default, and I'm not sure what module is currently used in the EL18.

nightflasher, 2024-02-02:
I'm not sure, if drivers needed? When someone sent all the telemetry to the UART the BT module is connected to, and I can scan for BT devices on my phone to find the NV14... At least telemetry should be able to get,without driver pain?!

juricabi, 2024-05-27:
Could this be implemented? https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/pull/28
Only for telemetry support.

richardclli, 2024-05-28:
It is just a serial passthrough BT module.  You can consider it is a wireless serial port after pairing.

juricabi, 2025-01-29:
> It is just a serial passthrough BT module. You can consider it as a wireless serial port after pairing.

That would be great if implemented, this is not implemented now in edgetx? Because i see only 'Nirvana' bluetooth, but cant pair.