EdgeTX/edgetx #702: Randomly bad channels value.

RipperGOT, 2021-09-10:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

pfeerick, 2021-09-13:
What difference does turning on or off the `ADC filter` setting on at the bottom of SYS/RADIO => Hardware make?

RipperGOT, 2021-09-14:
All tests are with the ADC Filter on, even when it has jitter. For some reason during reboots, sometimes it starts up with more, sometimes less, and sometimes nothing.
But only in the firmwares that I mention.
I think the factory firmware has a more "aggressive" ADC filter setting, I think ...   I'm not a programmer!  😅

If it was some of the hardware that disconnects the  ADC  filter , it would happen in all fimwares.

pfeerick, 2021-09-14:
I have heard mention of something about this before (jitter, and changing on reboot), so rest assured it's not just you! And I seem to think it was for a Radioking also that time.

rotorman, 2021-09-14:
Could this be an hardware issue of the analog supply or analog reference on TX18S? I do not have the schem of it to check, but if somebody does - maybe can check if TX18S analog rail might have coupling with digital or other rails?

pfeerick, 2021-09-14:
Ah, ha... it has hall effect gimbals, no? Thus the comment from here may apply... https://github.com/EdgeTX/edgetx/issues/648

RipperGOT, 2021-09-14:
Yes, I know I'm not the only one with this problem. And yes, they are hall effect gimbals.

I had also read as #648  indicates to turn off the ADC filter, but if I disconnect it, the jitter is bigger.

I have read in another # *** (I do not remember) this problem, but it is closed simply saying that it is a hardware problem.

I still think that it is not a hardware problem, because it is not replicated in all firmwares.

I think it makes sense and logic, right?

raphaelcoeffic, 2021-09-14:
> Yes, I know I'm not the only one with this problem. And yes, they are hall effect gimbals.
>
> I had also read as #648 indicates to turn off the ADC filter, but if I disconnect it, the jitter is [...]

The question is: which secret sauce has been used with the Radioking firmware ;-) Normally, they MUST publish it to you, the user. If they don't, they are in breach of the GPL license, which they probably don't care about, as usual. Forget about enforcing it in China....

RipperGOT, 2021-09-14:
Ok, I know what you mean. Chinese revealing secrets...  Ha!!
Yes, I also know who the manufacturer is, and that he did not publish the signature.

Ohhh man, I hope someone is kind enough to create some configuration with what you mention.
I would be happy to try it and give feedback. If I had enough programming knowledge, I would do it myself and upload it.
But I'm a donkey with this.

RipperGOT, 2021-09-15:
Sorry friends. Can i open pull request of these?
As I am not the only person with this problem, perhaps a specific configuration for Radioking could be implemented.
Test some settings and provide feedback.

rotorman, 2021-09-15:
> Can i open pull request of these?

If you have a solution, then a pull request would be very nice! (Pull request is the code snippet with the fix).

RipperGOT, 2021-09-15:
Okkkk.Thank you very much! I thought it was to ask for a change !!!!

Fu**** noob...   😅

pfeerick, 2021-09-15:
That's what issues are for - as well as bug and feature requests. But before anything can be done, it needs to be reproducible on multiple hardware, and we'd need to know or be able to figure out what Radioking are doing differently in their FW.

Steini63, 2021-09-28:
Maximum jitter on mine is 3us  (1501us - 1504us) at CH03. 

What i wonder about is that the "channel monitor" is slightly different from the "analog tests".
E.g. in CH04 is stable at 1% (1503us-1504us), while the same channel in the "analog tests" is stable at 0% (1283-1284 digits).
Maybe you should analyze the "analog tests" page.

richardclli, 2021-09-30:
This issue is related to #280

richardclli, 2021-09-30:
I do not think a poor hardware design can be solved properly by software alone.  So if you own a Jumper/Radioking hall gimbal, you cannot get rid of jitters.

The jitters comes from the hardware: 1. poor range design of hall gimbal (have only 1000 out of 4096 in precision), 2. mainboard interference (aka well known noise from speakers), 3. The 5 in 1 (that is why they downgrade to 4 in 1 afterwards)

Well, you can use custom code to apply a filter so that the problem is not VISIBLE only.  However, you will encounter delays when you do the real flight, so if you do not feel anything during flight, well leave the jitter as is, it is no harm.

If you really want to get rid of the jitter, change the radio or the gimbals.  The Flysky gimbals is the best out of the market.

RipperGOT, 2021-09-30:
> I do not think a poor hardware design can be solved properly by software alone. So if you own a Jumper/Radioking hall gimbal, you cannot get rid of jitters.
>
> The jitters comes from the hardware: 1. [...]

Hello friend, I think you have not read well, or you have not read the whole post. Since I already mentioned # 208

Ok, it has jitter and I accept it. But why not always? If there is a poor hardware problem (which I do not deny), it would always be!
This problem also does not occur so much with Opentx 2.3.14 (This version is not specific for RK and does not have the masking that you say.)

Why with Edgetx I start sometimes with 0 jitter and sometimes with a lot?
That is really the problem!

sfjuocekr, 2021-10-02:
I have tried to contact RadioKing about this to release their modifications, but got this reply and after inquiring multiple times I got zero response.

"Hi,
Thanks for your support.
I will feedback to the tech team."

My gimbals do jitter a bit, but the amount is very minimal.

pfeerick, 2021-10-03:
Ouch... it's like "Your feedback is important to us" (aka it went in the bin) :-O

RipperGOT, 2021-10-03:
@sfjuocekr  Hahaha sons of a bi...   I laugh for not crying!!!!  Where contact with they??? The web page is closed! 

@pfeerick From my ignorance, I have seen this code for RK support for OpenTX, I have been reading it and I think it will not have much impact on the radio, but could you take a look? 

[https://github.com/opentx/opentx/pull/8614/files](url)

pfeerick, 2021-10-04:
Yeah, it's 98% cosmetics. The only thing I see that set the added TX18S definition apart from the T18 definition apart is the setting of 
```
    set(HARDWARE_TOUCH YES)
    set(USB_CHARGER YES)
```
in the makefile... everything else is simply geared around making it so there is a TX18S compile entry, nothing more. It's a pity the OP just closed that PR rather respond, as it was a good PR in the fact it seems like it updated the firmware, companion and build system to the existence of the RK TX18S.

sfjuocekr, 2021-10-30:
Sorry, I just never got around to fixing the PR :(

Might pick it up again, I've had some struggles to deal with and still need to deal with.

Brownsy67, 2022-02-13:
> I have correct reads, but, turn off and turn on again and the reads are wrong. Make the same again and te reads are good... Its very randomly.
> 
> I have to reboot 4 or 5 times to get it to work fine !!!!
> 
> Ignore channel 2, throttle !(im in mode1) It's in ch 1, and 4
> 
> Good values.
> 
> [image]
> 
> Bad values [image]
> 
> Sorry, make the photo in the correct moment it's difficult. The values ​​changes from 1492 to 1500. Especially channel 4 and 1
> 
> Maybe read calibration in boot?
> 
> Radioking tx18s. EdgeTX 2.4.0
> 
> With opentx don't happen.
> 
> And sorry for my english
> 
> ----------------------------------UPDATE--------------------------------------------------
> 
> Opentx 2.3.10 Original from the Radioking--------- 0us jitter
> 
> Opentx 2.3.14-----------------------------------------1us or 2us max jitter
> 
> Edgetx 2.4.0-------------------------------------------8us or peaks 16us
> 
> Ok, the hardware sings a bit. But it doesn't seem like a hardware problem. With the firmware that comes from the factory it does not have any noise.
> 
> Problems with boot sequence? ADC settings different?
> 
> Maybe a modified factory setting for Radioking ??? Highest ADC filter setting ????

Could I get you to try something out for me?  I have recently run into this problem with the jitter on all four gimbal channels, but I no longer have the original firmware to test this on or revert back to.

I have read that the Jumper T18 has an issue with the speaker sounding crackly when the screen brightness is below 70%, since it goes without saying that these TX18S transmitters are manufactured by Jumper, and I notice the same speaker crackle issue in my TX18S. Raising the screen brightness to 100% seems to remove the speaker crackle, but it also removed almost 100% of the jitter in the gimbal sensors as well as channel readings on in Betaflight without having the ADC turned on.

I am curious if you or anyone else could do the same test, if possible with both a Jumper T18 and a Radioking TX18S, then report back the results.

It does seem odd that I had the same experience with the opentx firmware that it came with being spot on with the gimbal readings, but when updating to EdgeTX Jumper T18 firmware I get a whole mess of unflyable jitter.

This could be a hardware issue considering that changing the screen brightness does seem to improve the problem, and that there is noise making its way to either the hall effect sensor itself or the main IC.  But it does not seem to make sense to me that this issue was not present with the original "hacked together" OpenTx firmware that came installed on the radio.

There has got to be a solution to this either in the code or with the hardware. I'm really hoping there is a bit of code that some smarter person could tweak to make this work, but if not I'm sure that this may be able to be solved with a bit of filtering of the power rails.

Thank you for taking the time to read the wall of text. I am willing to help anyway I can with this issue, as long as it doesn't involve writing code. I know nothing of that aside from basic arduino coding. I own an oscilliscope, and I am willing to probe around in my TX to help find a solution.

Brownsy67, 2022-02-14:
If at all possible, could I please get a copy of the original firmware from someone please. No luck finding it anywhere.

pfeerick, 2022-02-14:
Is that the stock firmware for the RK TX18S you're after? I wish you luck, as it may only be someone who still has it on their transmitter and can read it off via Companion (i.e. firmware backup) or DFU who can supply it.

RipperGOT, 2022-02-15:
@Brownsy67 You are a lucky boy! I have the firm!
[Firm-Original.zip](https://github.com/EdgeTX/edgetx/files/8065092/Firm-Original.zip)

Another matter, I gave this radio to my father and I have not been able to dedicate much time to it. He uses the original firm zero problems OTX 2.3.10.

This week I'll pick it up and take a look at it with Edge TX 2.6. 

Sorry for my english, google translate

Brownsy67, 2022-02-15:
RipperGOT, Thank you!! A lot. I will be doing extensive testing to find a solution on the hardware side of things between Edgetx and the original shipped Opentx 2.3.10.

I am new to github and still unsure how to properly use the site. Where is best to post screencaps of oscilloscope readings, here in this post or someplace else?

EdgeTX 2.6 - I am getting the same results as my above post when I turn off the alarms, turn off the sounds, turn off the haptic¿? alarm, lower all volumes, and having the screen brightness (both sliders) all the way down. I am noticing that if the jitter is more than +/- 1% (ADC off) a power cycle will bring it back to almost dead on centre, with a very tiny jitter between 0 and +/- 1% when sticks are centered as well as when moved from center. Much much more acceptable already at this point. 

still working out in my head the best way to tackle this on the hardware side with filtering of the power rails of the main IC, of the Vref for the hall effect sensors, or maybe it's something as simple as pulling the cmos battery for a few seconds and replacing it. Still have yet to tear this thing open and poke around properly, trying the non invasive stuff first.

Thank you RipperGOT for providing the original factory firmware for me to test with, and to anybody else that is helping to find a solution for this (not so terrible) radio.

rotorman, 2022-02-15:
You can include the screenshots/scope pictures here in GitHub Issue post, then all is in one place (just drag or paste the image file here in the text field).

About the jitter - try disconnecting the 12-pin internal RF-module cable to see if the jitter is gone when no RF-module is connected? Not a realistic scenario for controlling a model w/o RF, but that test will tell if the issue might stem from RF-module interfering with the analogs on TX18S.

Brownsy67, 2022-02-24:
I have checked numerous times so far with both the original firmware as well as EdgeTx 2.5 and 2.6. All 3 have been tested with and without the RF module connected, and all still show the same jitter on the gimbal outputs in both scenarios. The difference I have noticed so far with just messing about is that with both versions of EdgeTX my sticks seems to fall out of calibration all together after about 10-15 power cycles. I dont know if this is significant at all, but I have noticed this. 

I still have not ruled out filtering the power rails to solve this, still working down the list.

gnuton, 2023-03-04:
Is there any update about this? I do have a jumper t18s with latest edge tx and i do notice jitter on all potentiometers not just the gimbals. I fill this is an hardware issue.

macem, 2023-07-07:
I already bought RadioKing T18S, updated to EdgeTX 2.6 and I have the same problem. When I calibrate the sticks everything works fine until I restart. Then I see that jitter on some channels :(. Really strange behavior.

gnuton, 2023-07-07:
@macem do you see improvements rolling back?