EdgeTX/edgetx #708: Reset ALL function missing

yds, 2021-09-11:
[image]

there's no option to select **All** as the target of the Reset -- very useful to be able to have a single SF to reset all Timers and Sensors which need to be zeroed before next flight.

The 4 char width cutoff for the menu options seems like a bug when displaying the Timers to reset.

pfeerick, 2021-09-18:
Hm, this is a split issue... UI glitch on the NV14, and missing "All" reset option...

pfeerick, 2021-09-18:
@yds I may have misunderstood this... is by "All", did you mean the "All" that is shown on the B&W radios - referred to as "Flight" on colour screen radios - or something different?

yds, 2021-09-18:
@pfeerick I guess I meant "All" as in B&W radios.  is that now called "Flight" on color screen radios?

I was trying to recreate the model I had set up under FlySky-OpenTX where "All" was one of the options when selecting the Reset function -- see SF22 at the bottom of the screenshot:

[image]

raphaelcoeffic, 2021-09-18:
> @pfeerick I guess I meant "All" as in B&W radios. is that now called "Flight" on color screen radios?
> 
> I was trying to recreate the model I had set up under FlySky-OpenTX where "All" was one of the options when selecting the Reset function

yes, this is called “Flight” on color radios.

closed 2021-09-18 by pfeerick