EdgeTX/edgetx #711: Higher baud rates #640 with TBS-Crossfire issue

zyren, 2021-09-12:
The latest submission Higher baud rates #640, I tested it on TX16S, except 115200 other baud rates can not use TBS Crossfire, and at 115200, crossfire configure can only get the first half of the menu, can not be displayed completely, use TBS Agent Lite this lua script can get a complete menu.
Crossfire firmware has been upgraded to V6.09

pfeerick, 2021-09-12:
What happens when you power cycle the radio after changing the baud rate? I just tried the latest nightly which has the PR included, and the ELRS lua also doesn't seems to respond after changing the baud rate, but if I restart the radio it seems to be fine.

zyren, 2021-09-12:
> What happens when you power cycle the radio after changing the baud rate? I just tried the latest nightly which has the PR included, and the ELRS lua also doesn't seems to respond after changing the baud rate, but if I restart the radio it seems to be fine.

But Crossfire cannot be used no matter how it restarts.

raphaelcoeffic, 2021-09-12:
@zyren is that only the script, or is there an issue as well with the module getting the stick values?

zyren, 2021-09-12:
> @zyren is that only the script, or is there an issue as well with the module getting the stick values?

Only at 115200, it can be used barely. The receiver responds normally, but at 400k and other baud rates, it cannot be used normally. There is no Telemetry, and the receiver does not respond. I think it may be that the synchronization of CRSF 400k bps may have deviated. Using yesterday's firmware, you can get RC Input: CRSF Shot very well.

raphaelcoeffic, 2021-09-12:
@pkendall64 did you notice anything? Seems to break TBS Crossfire.

pkendall64, 2021-09-12:
With my TX16S I can use both ELRS LUA and Agent Lite with my ELRS module all the way from 115k to 3.75M with 921k not working.

pkendall64, 2021-09-13:
@zyren Could you test this patched version with your crossfire unit. This version has a patch that only enabled 25Mhz GPIO and 8bit sampling on UART if over 400k baud is selected.
I do not think your crossfire will work with the higher baud rates, so only 115k and 400k should work.
 
[tx16s-5dd4a89eb.bin.zip](https://github.com/EdgeTX/edgetx/files/7151296/tx16s-5dd4a89eb.bin.zip)

zyren, 2021-09-13:
> @zyren Could you test this patched version with your crossfire unit. This version has a patch that only enabled 25Mhz GPIO and 8bit sampling on UART if over 400k baud is selected.
> I do not think your crossfire will work with the higher baud rates, so only 115k and 400k should work. If this works for you, I will create a new PR with the changes.
> 
> [tx16s-5dd4a89eb.bin.zip](https://github.com/EdgeTX/edgetx/files/7151296/tx16s-5dd4a89eb.bin.zip)

Ok i test it now

zyren, 2021-09-13:
Great, this firmware works perfectly at 400kBps and is CRSFshot.
Thank you very much!

[image]
[image]
[image]

pkendall64, 2021-09-13:
Sweet, I'll make a new PR

PR #715 (2021-09-13): Enable medium GPIO speed and 8bit sampling on high baud rates only
This fixes #711 as verified by the user that raised the issue. Also continues to work on ELRS, on all baud rates.
closed 2021-09-13 by pfeerick