EdgeTX/edgetx #714: Keyboard doesn't call setChangeHandler directly after softkey enter

dlktdr, 2021-09-12:
I've been working on the simplified category editor and having an issue when the keyboard is used to rename on a TextEdit

If using the hardware keys only,
- You press enter on the TextEdit
- Change the string
- Press enter again
- setChangeHandler gets called immediately as expected. 

If you use the software keyboard.
- Touch the text edit
- Change the string
- Press Soft Key Enter
- Nothing...
- Change the focus, touch elsewhere or a hardware scroll. Then setChangeHandler gets called.

If you use the software keyboard and press Rtn instead of the softkey it works as expected.

It's causing an issue as if you wait for the user to press or move something else I can't update the group list properly. Haven't found the solution yet I'll keep looking but I know others @kevinkoenig knows this keyboard a lot better than I.

kevinkoenig, 2021-09-12:
I will look at it tomorrow morning.  I am sure there is an easy solution.

dlktdr, 2021-09-15:
Found a fix.. Yeah it was actually pretty easy. https://github.com/EdgeTX/libopenui/pull/22

closed 2021-09-15 by dlktdr

kevinkoenig, 2021-09-15:
This is a good temporary fix and should be merged.  We move changehandler to FormField() to enable this in other areas.  I have a fix in progress that solves the problem in that way, which we can leverage for other change management capabilities.

From: Cliff
Sent: Wednesday, September 15, 2021 4:21 PM
To: EdgeTX/edgetx
Cc: Kevin Koenig; Assign
Subject: Re: [EdgeTX/edgetx] Keyboard doesn't call setChangeHandler directlyafter softkey enter (#714)

Closed #714.