EdgeTX/edgetx #718: Scale field in the Inputs tab

MrVinther, 2021-09-13:
- EdgeTX 2.4.0 Flysky Hall gimbals
- Radiomaster TX16S

In the Inputs tab the field "Scale" will only go as high as 100.
Im trying to get it to read from the capacity telemetry RB1C as source with a scale up to 4500mah. This worked just fine in OTX as the first picture below. Second picture is EdgeTX which is stuck at 100.
Converted models from OTX works fine until I change the Scale value ever so slightly, jumps back to 100.

[image]
[image]

pfeerick, 2021-09-13:
That's OTX 2.4, isn't it? Looks like it may be a variable related error, since it jumps from 0 to 16383 then back to 100 😮 

@elecpower btw, whatever this "Scale" is, it's missing in Companion ;)

eshifri, 2021-09-15:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/model_inputs.cpp#L245
     
/* TODO telemetry current value
      if (ed->srcRaw >= MIXSRC_FIRST_TELEM) {
        drawSensorCustomValue(EXPO_ONE_2ND_COLUMN+75, y, (ed->srcRaw - MIXSRC_FIRST_TELEM)/3, convertTelemValue(ed->srcRaw - MIXSRC_FIRST_TELEM + 1, ed->scale), LEFT|(menuHorizontalPosition==1?attr:0));
      } */
      grid.nextLine();

      // Scale
      // TODO only displayed when source is telemetry + unfinished
      new StaticText(window, grid.getLabelSlot(), STR_SCALE, 0, COLOR_THEME_PRIMARY1);
      new NumberEdit(window, grid.getFieldSlot(), -100, 100, GET_SET_DEFAULT(line->scale));
      grid.nextLine();

      // Weight

PR #745 (2021-09-17): Support scale 
Support scale for telemetry sensor. Fix #718
closed 2021-09-17 by pfeerick