EdgeTX/edgetx #720: Windows build is broken

eshifri, 2021-09-14:
See title.

Booloader build fails with the error message:

[ 29%] Generating mask_topleft.lbm
mingw32-make[3]: Leaving directory 'D:/MyDownload/OpenTX_Code/edgeTX/WithGit/eshifri/edgetx/buildTX16S'
[ 29%] Built target x10_themes_masks
mingw32-make[2]: Leaving directory 'D:/MyDownload/OpenTX_Code/edgeTX/WithGit/eshifri/edgetx/buildTX16S'
CMakeFiles\Makefile2:222: recipe for target 'radio/src/CMakeFiles/firmware.dir/rule' failed
mingw32-make[1]: *** [radio/src/CMakeFiles/firmware.dir/rule] Error 2
mingw32-make[1]: Leaving directory 'D:/MyDownload/OpenTX_Code/edgeTX/WithGit/eshifri/edgetx/buildTX16S'
Makefile:156: recipe for target 'firmware' failed
mingw32-make: *** [firmware] Error 2

pfeerick, 2021-09-14:
`pacman -S mingw-w64-x86_64-clang` 

It was dropped during one of the documentation updates. And due to it being a multi-threaded build, the error is actually much further up... i.e. 
```
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.9/site-packages/clang/cindex.py", line 4173, in get_cindex_library
    library = cdll.LoadLibrary(self.get_filename())
  File "C:/msys64/mingw64/lib/python3.9/ctypes/__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "C:/msys64/mingw64/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'libclang.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```

eshifri, 2021-09-14:
I'm still using deprecated build chane. Any way to fix it there?

pfeerick, 2021-09-14:
Not sure what you mean... unless your error is different, the problem is that clang is missing, so needs installing, hence 
> `pacman -S mingw-w64-x86_64-clang`

raphaelcoeffic, 2021-09-14:
> I'm still using deprecated build chane. Any way to fix it there?

Are you using mingw32? Any reason for not using mingw64?

pfeerick, 2021-09-14:
Oh, nice catch ... `mingw32-make`

eshifri, 2021-09-14:
Yes, as I said I'm using original build system (marked as deprecated)  :-)
I will switch to new one.

closed 2021-09-14 by raphaelcoeffic

eshifri, 2021-09-15:
I know it was closed, but... :-)
I installed the MSYS2 and all the tools.
I can built TX16S.
I can built NV14 simulator library.
But NV14 firmware build fails with:

[  0%] Generating datacopy.cpp
./definitions.h:25:10: fatal error: 'stdbool.h' file not found
  datastructs.h:27:10: note: in file included from datastructs.h:27:
  targets/nv14/board.h:24:10: note: in file included from targets/nv14/board.h:24:
make[3]: *** [radio/src/CMakeFiles/datacopy.dir/build.make:74: radio/src/datacopy.cpp] Error 127

Obviously stdbool.h is present in the mingw.
Any ideas?
Thank you!

raphaelcoeffic, 2021-09-15:
> I know it was closed, but... :-)
> I installed the MSYS2 and all the tools.
> I can built TX16S.
> I can built NV14 simulator library.
> But NV14 firmware build fails with:
>
> [ 0%] Generating datacopy.cpp [...]

This happens when the datacopy target cannot grab the gcc headers (have a look at the macro just above the target definition). Last it was because the terminal was using non-English locales.

eshifri, 2021-09-15:
Thank you! This was helpful.
datacopy.cpp was not created for NV14 target. After I have copied it from TX16S, the build went through.

raphaelcoeffic, 2021-09-15:
> Thank you! This was helpful.
> datacopy.cpp was not created for NV14 target. After I have copied it from TX16S, the build went through.

It should created on the fly, please check the command by running make with `VERBOSE=1`.

eshifri, 2021-09-16:
CMakeLists.txt for NV14 calls generate_datacopy.py with ${arm_gcc_include_path}.
This variable is not defined in this file.

Similar file for horus uses ${gcc_include_path}
which is defined in both NV14 and horus versions.

raphaelcoeffic, 2021-09-16:
> CMakeLists.txt for NV14 calls generate_datacopy.py with ${arm_gcc_include_path}.
> This variable is not defined in this file.
> 
> Similar file for horus uses ${gcc_include_path}
> which is defined in both NV14 and horus versions.

Oh! I see! I probably forgot to change it! Just wondering how that works for me here and on GH actions.... could you please change it and retest? (+PR if that works?)

pfeerick, 2021-09-16:
This is something specific to the msys2 mingw build process - and doesn't occur under the docker build environment - hence why it's working everywhere else. I can replicate this also on msys2-mingw, but normal build environment is fine. Plus I don't use msys2-mingw for firmware builds as it's about three/four times slower than docker/WSL2.

raphaelcoeffic, 2021-09-16:
> This is something specific to the msys2 mingw build process - and doesn't occur under the docker build environment - hence why it's working everywhere else. I can replicate this also on msys2-mingw, [...]

I see! I don't need these extra headers when compiling natively on OSX, as I have a full install of clang that relies on its own set of headers. MSYS has clang only using GCC headers though....