EdgeTX/edgetx #722: Output > 101.3% or < -101.3% switches to GV

pfeerick, 2021-09-14:
There are two reports on Discord about this

> hmm, so on my TX16S on build #542213de, if an output is more than 101.3% or less than -101.3%, it turns into a GV instead of -101.3% or something

> same for me with extended limits on it puts them into the GV FW version 2.5.0 (f5c946a0)

@eshifri Any ideas, since it probably stems from #687

eshifri, 2021-09-14:
I'm sure it is related to #687: everything above 101.3% is treated as a GV. Or above 15X for extended limits.
Should we limit the MIN and MAX in numberEdit?

pfeerick, 2021-09-14:
Hm, I think it's correctly limited already using GVarNumberEdit, and I can't actually reproduce this myself at the moment... output will only go to min -100, max +100 (-150,+150 for extended limits) and even with EL on outputs screen is still saying everything is fine... 542213d is yesterdays nightly and am running that and everything 'seems' fine? This is plain weird... need more info on how to reproduce this...

rburrow87, 2021-09-14:
Posted this on Discord already but here's what I'm seeing:

Build 88b8ac86 where it's all good:
[image]

Build 542213d where it goes wrong:
[image]

I see this happen on a freshly created model under 542213d too. 88b8ac86 was just the last nightly I had saved, not sure which nightly this appeared.

eshifri, 2021-09-14:
This happens with EL and it is deeply rooted in the way GVs are distinguished from values.
The culprit is in the PREC1 attribute used for limits and offsets.

jmxp69, 2021-09-14:
I can confirm same issue from my compiled version. Using extended limits and outputs > 100, when I leave the outputs page and go back, the output is changed to a gvar.

pfeerick, 2021-09-14:
Thanks for the report. I finally realised the trick to reproducing was exiting the page, so have been able to test the fix being worked on. Preminimary rltesring shows its working, and just the cosmetics fixed now.

pfeerick, 2021-09-15:
#726 fixes this for me, so please yell out if something still isn't right.

PR #726 (2021-09-15): Call GV_IS_GV_VALUE() without offset applied
Resolves #722
commit ff0abf11 (2021-09-15): Call GV_IS_GV_VALUE() without offset applied

Resolves #722
closed 2021-09-15 by pfeerick