EdgeTX/edgetx #724: Battery Bar Indicator Shifted Left

jmxp69, 2021-09-14:
On the top bar, the battery bar indicator does not ever show the final bar even with correct calibration. It looks like the vertical bar set is shifted one row to the left. Confirmed by multiple users on multiple versions. I'm using 2.5.0 (69efe13a) with flyskyhallsticks from Risto.

rburrow87, 2021-09-14:
Picture for reference. This is with the normal nightly 6278335e but has been present for a while now. Not sure which one it started in.

[image]

rotorman, 2021-09-14:
Here on the picture 5 battery bars are shown, but drawn unfortunately a bit too much to the left.

rotorman, 2021-09-14:
Fixed by 1eed24c
Old:
[image]
Fixed:
[image]

closed 2021-09-14 by rotorman

pfeerick, 2021-09-14:
Lol... Oh my, that is one deceptive little bugger, looking like the battery symbol had positive on the left side! Nice one!