EdgeTX/edgetx #729: The problem of adjusting the OUTPUTS extend limit

zyren, 2021-09-15:
Firmware version: newest nightly

The problem of adjusting the OUTPUTS extend limit, if the min or max adjustment exceeds -100 or +100, it will automatically become the wrong GV value and cause the channel to become unresponsive.

[image]
[image]

rburrow87, 2021-09-15:
Which build are you on? This just got fixed today and is in the current nightly release.

pfeerick, 2021-09-15:
Double check your version - if it doesn't say `0dac6037` as the commit after the version you're probably running yesterdays nightly still, and I can't reproduce this any more.

zyren, 2021-09-15:
> Double check your version - if it doesn't say `0dac6037` as the commit after the version you're probably running yesterdays nightly still, and I can't reproduce this any more.

Great, I am using yesterday’s nightly version, I will update to today’s latest version, thank you very much!

closed 2021-09-15 by zyren