EdgeTX/edgetx #730: Color element tweaks

pfeerick, 2021-09-15:
@JimB40 This one's for you... so you'd better have ideas on what needs fixing! 😝 

- [ ] 1. Presumably change icon colours to white, rather than a theme colour? Otherwise, they could probably be made the same element colour used for the top bar?
[image]

- [ ] 2. What else needs tweaking... screenshot with hilights/circles and preferably a hint as to what colour element it should be mapped to would be good ;)

JimB40, 2021-09-16:
Doing review of other elements. Will post soon screenshots

JimB40, 2021-09-16:
[image]
Okay so first little changed Testing color scheme

JimB40, 2021-09-16:
1. Wrong label text color. 
They Should be PRIMARY1 (P1) 
Two cases
a. like on **Main view 1** placed on background (field desctiption)
b. like on **LOGICAL SWITCHES** etc text label when LS is not set. We can see there is no value set so no need to dim label text 
Below places I've spotted to make this change 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

JimB40, 2021-09-16:
FLIGHT MODES
Should be like other screens with ediatabe fields
Left labels - background S2, text P1
left labels focused - background FO, text P2

[image]

JimB40, 2021-09-16:
GLOBAL VARIABLES
Left labels - background S2, text P1
Values text color S2 (not S1)
plus all 9 FM varv should fit to one line that makes it more readable. (you can make labels column GV1, GV2 smaller)

[image]

JimB40, 2021-09-16:
QUICK MENU
That's simple 
1. Make it Black & White :)
2. Set opacity of background to 70-80% so it's better readable in light enviroinment.

[image]

JimB40, 2021-09-16:
SD CARD
Ditch gray background

[image]

JimB40, 2021-09-16:
POP-UP Mask 
It should be BLACK. Alternatively we can try S3 (main background color) but with light backround schemes it may not work

[image]

closed 2021-09-29 by pfeerick