EdgeTX/edgetx #731: Telemetry/Input lists showing everything

rburrow87, 2021-09-15:
TX16S nightly 0dac6037 build

When going into a list of inputs and telemetry sensors and whatnot, everything possible is being listed instead of only what is set up.

For example, there are a bunch of blank telemetry sensors:

[image]

Also seeing L01-L64 despite only a couple being set up, inputs that are not set up, switches/sliders/knobs that don't exist, etc.

pfeerick, 2021-09-15:
Yes, it does seem to ahve gone a bit crazy. Also a couple of `[--]` entires between 'GPS' and 'Flight'.

wimalopaan, 2021-09-15:
This is the same for Mixer-scripts :-(

jmxp69, 2021-09-15:
Same behavior here. Something to do with the GVAR Outputs bug just fixed?

kevinkoenig, 2021-09-15:
Could this be related to the change made to reimplement sourcechoice recently

PR #739 (2021-09-16): [color] properly filter sources in drop-downs
Resolves #731
commit cfdee7f0 (2021-09-16): [color] properly filter sources in drop-downs

Resolves #731
closed 2021-09-16 by raphaelcoeffic