EdgeTX/edgetx #732: List of Inputs shows everything

wimalopaan, 2021-09-15:
The list of inputs of e.g. mixer shows every possible lua mixer outputs. This is in simu25, companion25 and radio:

[image]

wimalopaan, 2021-09-15:
Looks like connected to #731

wimalopaan, 2021-09-16:
[image]

This shows all inputs up to 32 which are not defined

wimalopaan, 2021-09-16:
[image]

This shows the mixer scripts: the first line is an existing mixer-script-output, the other lines are non existing mixer outputs (the also have an icon in front of them).

philmoz, 2026-02-09:
This can probably be closed.