EdgeTX/edgetx #733: Cannot use the scroll wheel to modify the Refresh rate of SBUS on External RF

zyren, 2021-09-15:
When SBUS is selected for External RF, the Refresh rate cannot be modified with the scroll wheel, and the virtual keyboard can be modified with << and >>, but it cannot be modified with +/-.
[image]

pfeerick, 2021-09-15:
What about if you scroll the wheel fast? I can reproduce similar behaviour non-changing if using -/+ on the touch keyboard, but `<<` and `>>` work, probably because they increase the next significant digit. It seems like at certain numbers, you need to scroll the wheel faster to get it to move, but should change (TX16S).

zyren, 2021-09-15:
> What about if you scroll the wheel fast? I can reproduce similar behaviour non-changing if using -/+ on the touch keyboard, but `<<` and `>>` work, probably because they increase the next [...]
Oh, it is indeed necessary to roll faster, slow speed does not work, the roller I am using is modified according to Risto's enhancement roller board, and all functions are currently normal.
Thank u!

pfeerick, 2021-09-15:
Excellent... at least it's a easily reproducible bug! We like those, usually easier to squash!

pfeerick, 2022-03-22:
This was resolved in #864

closed 2022-03-22 by pfeerick