EdgeTX/edgetx #734: Companion Application Settings says OpenTX

timtuxworth, 2021-09-15:
**Describe the bug**
Settings Application Settings says "Automatic check for OpenTX firmware updates" should probably say "EdgeTX"

**To Reproduce**
Start EdgeTX companion - click settings button or click Companion TX -> Preferences
Click Application Settings tab

**Expected behavior**
Prompt should read Automatic check for EdgeTX firmware updates"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 Companion 2.5 Sept 15 2021

pfeerick, 2021-09-15:
:laughing: There are plenty of places that need the name changing still, including things like the simulator filenames, but thank you for catching this one!

raphaelcoeffic, 2021-09-16:
> 😆 There are plenty of places that need the name changing still, including things like the simulator filenames, but thank you for catching this one!

I think we should disable anything related to firmware download for now, as it is not functional. (@elecpower do you think this can be done easily?)

pfeerick, 2021-09-16:
If we were really lazy, it could even simply be a matter of hiding the controls from view... it might look a little "odd" but at least nothing would break and can be turned back on again easily.

raphaelcoeffic, 2021-09-16:
> If we were really lazy, it could even simply be a matter of hiding the controls from view... it might look a little "odd" but at least nothing would break and can be turned back on again easily.

Fine for me, whatever works.

pfeerick, 2021-09-16:
Is this enough? As I can push a PR for that now. Neil will need to do the last part if you want more, as it looks like hiding may not be as straightforward as I thought since a quick search for that goes straight to stackoverflow rather than a nice QT documentation page. 😁 

[image]

timtuxworth, 2021-09-16:
This looks good to me. I'd say disable rather than hide is better if its intended to be functional eventually. Thanks for your efforts guys!

elecpower, 2021-09-16:
How to hide noted on PR

PR #737 (2021-09-16): Tweaks to companion/simulator 
Resolves #734

Also tidies up a couple more references to OpenTX
closed 2021-09-16 by pfeerick