EdgeTX/edgetx #735: Suggest more convenient copy trims to subtrim

zyren, 2021-09-16:
In the OUTPUTS page, after Copy trims to subtrim, please reset the trim, otherwise the subtrim in the OUTPUTS project will overlap with the trim. In addition, it is recommended to add a one-click copy of all trims to the all channels Subtrim at the bottom of the outputs page, which will be more convenient.
In the SPECIAL FUNCTIONS page, setting Inst.Trim can also complete this function, but I think it will be more convenient to select in the OUTPUTS page.

pfeerick, 2021-09-16:
Duplicate of https://github.com/EdgeTX/edgetx/issues/354

pfeerick, 2021-09-16:
I feel the individual copy function should stay as it is, otherwise it should be renamed as well as behaviour changed (since it is "copy" not "move"). And as mentioned in the other issue, it is missing, and needs adding back.

zyren, 2021-09-16:
Ah, that's right, the meaning of copying can be misleading.
However, the bottom Trims => Subtrims is very easy to use, and when Trims move to Subtrims, the original trims will be reset to the center position.
[image]

pfeerick, 2021-09-16:
Yes, exactly, which is already covered in #354 ;) It is planned for the option for move trims to subtrim to be re-added, and most likely added to the top so it's more visible as well.

closed 2021-09-16 by pfeerick

zyren, 2021-09-16:
Wow, I’m sorry I didn’t carefully check the content submitted before, the same proposal is repeated, thanks for your tip！

pfeerick, 2021-09-16:
No problem, it just means more reason to get it implemented ;)