EdgeTX/edgetx #736: Switch Hardware List Out of Sequence

jmxp69, 2021-09-16:
Hardware list isn't quite right. 
[image]

[image]

raphaelcoeffic, 2021-09-16:
@jmxp69 I'd bet this caused by the `flyskyhallsticks` ;-) Can you please have a look without that option?

pfeerick, 2021-09-16:
I would suggest so also... I'm running essentially a nightly build and without `flyskyhallsticks` and the list is right.

rotorman, 2021-09-16:
Indeed, an issue of FLYSKY_HALL_STICKS option. Here without FLYSKY_HALL_STICKS build option:
[image]
and here with:
[image]
Now need to find the line in code where the issue stems from...

rotorman, 2021-09-16:
@jmxp69 Hi John - can you please test this fix out?
As I know, you are liking self-compling, to test this PR, you need to switch to this PR branch, as the code is not yet merged in main (thus just running `git pull` won't fetch you the update). Here are the steps for MSYS2:
```
cd ~/edgetx/edgetx_main
git pull
git checkout EX3_EX4_Bugfix
```

To check that you are in correct branch, run `git branch` - the EX3_EX4_Bugfix should be highlighted.

Then clean the build output dir and rerun the std. CMake and make:
```
cd build_output
rm -rf *
cmake -G "MSYS Makefiles" -DCMAKE_PREFIX_PATH=$HOME/5.12.9/mingw73_64 -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DLUA=YES -DFLYSKY_HALL_STICKS=YES -DCMAKE_BUILD_TYPE=Release ../
make -j`nproc` firmware
```

Thanks!

jmxp69, 2021-09-16:
Looks good to me. The switches appear correctly in mixer now too. Thanks guys. 

@rotorman git branch does show EX3_EX4 fix highlighted. Should I switch back to main?  Is it git checkout main?

Thank you.

[image]
[image]

FYI: I found this today--makes posting screenshots to GH a little easier. https://github.com/Tichau/FileConverter/releases/tag/v1.2.3

rotorman, 2021-09-16:
@jmxp69 John - thank you for testing. The code is merged now, thus yes, please switch back to main branch:
```
git checkout main
git branch
git pull
```
Best,
Risto

closed 2021-09-16 by rotorman

jmxp69, 2021-09-16:
Thanks for the fix and help with checking the updates out. Appreciate you all. You're doing amazing work.

pfeerick, 2021-09-16:
If it's `simu`/simulator, as Risto said, just drag and drop as GitHub supports PNG. You can even set Simulator to copy to clipboard instead, so no pesky file if you just want to share it. If it's `.BMP` screenshots from the radio, The simplest way for me on Windows is to (right click, Open With) the "Snip and Sketch" tool  and then just copy to clipboard and paste to GitHub, no file conversion necessary. That looks like a handy tool though.

jmxp69, 2021-09-17:
These are radio screenshots. I didn't check Companion for the error.

PR #743 (2021-09-16): Fixes EX3 and EX4 bug caused by myself
Resolves #736 when FlySky hall sticks are built into firmware.

Before this bugfix:
[image]

After this bugfix:
[image]