EdgeTX/edgetx #750: Animated tree diagram that shows how each output is derived from inputs through mixes, outputs and overrides

timtuxworth, 2021-09-17:
On the fancy new radios (like my new TX16S that is still in it's box), could we have a graphical display that how's how an output is derived? I have a picture in my mind of a tree that shows the input through to the mixes, outputs and even overrides that shows what happens at each step and updates as a stick is moved or a switch is set. 

This could also be a page in the simulator for those radios that don't have the big display.

This came up in a discussion on OpenTX about channel overrides.
https://github.com/opentx/opentx/issues/8701#issuecomment-921862033

For new users, or occasional users, it can be very hard to visualize or figure out exactly what is going on when the model does something unexpected based on a particular setup.

could we have a display that how's how an out put is derived? I have a picture in my mind of a tree that shows the input through to the mixes, outputs and even overrides that shows what happens at each step and updates as a stick is moved or a switch is set. This could also be a page in the simulator for those radios that don't have the big display.

Alternatives - draw it out on a. piece of paper?

See the above link to the OpenTX discussion for context. This is a bit like an uber version of the line highlighting that you see in the mixes page on the radio when swtches are set.

timtuxworth, 2021-09-17:
This is a suggestion for an enhancement. I can't figure out how to add the enhancement tag.

rotorman, 2021-09-17:
@timtuxworth This sounds like a nice article for Wiki instead to me? If you would prepare a page (any format of your liking), I could convert it to a markdown and put it up in the EdgeTX non-dev Wiki.

timtuxworth, 2021-09-17:
Where do I find the non-dev wiki @rotorman ? I'm ok writing markup.

timtuxworth, 2021-09-17:
I guess I need to draw a picture. I'll see what I can come up with.

pfeerick, 2021-09-17:
@rotorman perhaps you mean Discussion, not wiki/documentation, since this is a proposal for a feature, not something existing that needs documentation? Or have I missed something?

rotorman, 2021-09-17:
@pfeerick maybe I misunderstood indeed.
I gathered that the desire was for a diagram of how the values from inputs are passed through various functions. @timtuxworth - maybe you can please confirm if a static help page would be sufficient or if this needs to be a dynamic on the fly generated diagram on the radio? Thx!

timtuxworth, 2021-09-17:
Oh no. I'm suggesting an "animated" diagram on the radio that is updated dynamically in real time as the user moves the sticker and switches the switches and pots.

gagarinlg, 2023-03-18:
@philmoz looking for a challenge 😉

philmoz, 2023-03-18:
> @philmoz looking for a challenge 😉

TBH I have no clue what this would even look like. If someone wants to mock up a design I might be able to code it.

rotorman, 2023-03-18:
@timtuxworth would you mind putting your idea into one or more drawings to give us further clues of how the end result should look like?