EdgeTX/edgetx #752: Replacement icon masks

pfeerick, 2021-09-18:
Replacement icons are needed for three icons used on the fullscreen dialogs, and are located in `/radio/src/bitmaps/480x272/default_theme`  (https://github.com/EdgeTX/edgetx/tree/main/radio/src/bitmaps/480x272/default_theme)

Specifically, 
- [ ] mask_asterisk.png
- [ ] mask_busy.png
- [ ] mask_question.png

But also feel free to update the following two if so desired. 
- mask_error.png
- mask_shutdown.png

Only requirement is that they must be black and white since they are masks. All but the last image are 105x105, with that latter being 96x96.

JimB40, 2021-09-18:
@pfeerick I’ll do it in exchange for #638 ;)

pfeerick, 2021-09-19:
Either 105x105 for first four, and 96x96 for last, meaning they are a drop in replacement that can go in any time, or we lay this over for 2.6 and look into tidying up those dialog screens, and probably make them all 96x96 or smaller if we can get away with it for consistency... will just have to keep in mind smaller also equals harder to see at a glance when out at the field in the sun...

JimB40, 2021-09-19:
Do we go 96x96px for all of them then. I can enlarge icon to fill 96px canvas entirely

pfeerick, 2021-09-19:
Nothing is jumping out at me saying they need to be a specific size as it looks only the top left corner is specified, no width or height... so lets go for 96x96px for consistency and the few extra bytes of space saved ;)

JimB40, 2021-09-20:
You're welcome

[image]
[image]
[image]
[image]
[image]

JimB40, 2021-09-20:
Here is ZIP as GH changes names :)
[ETX_SYS_icons.zip](https://github.com/EdgeTX/edgetx/files/7193876/ETX_SYS_icons.zip)

commit 4e7b18f0 (2021-09-20): Replacement icon masks thanks to @JimB40
Resolves #752
PR #770 (2021-09-20): Replacement icon masks thanks to @JimB40
Resolves #752

Passed tests locally, but lets see what GH thinks... Nice.
Now for transmitter checks :)

[image]
[image]
[image]
commit e51ffb11 (2021-09-20): Replacement icon masks thanks to @JimB40
Resolves #752
closed 2021-09-20 by pfeerick