EdgeTX/edgetx #758: Inputs Source Max in Companion is Changed During Write to Radio

jmxp69, 2021-09-18:
2.5.0 (514796fa) flyskyhallsticks, noheli + standard make options

When setting source to MAX in an Input on Companion, after writing models/settings to radio, the source is changed to "0" in the Source line of the input editor. It shows as JSx (Pot) in the Inputs list. When I change from JSx (Pot) to a switch like SB, the radio reverts to Emergency Mode. This is repeatable. I can change to 

[image]

When I read the same model from radio into Companion, the source shown is CYC.

jmxp69, 2021-09-19:
https://youtu.be/rGhFryQ_tC0

raphaelcoeffic, 2021-09-19:
@jmxp69 does this include any radio interaction? what do you mean with "after writing models/settings to radio"?
Regarding the source being different on radio and companion, this could be caused by the extra sources added for the `flyskyhallsticks`, which Companion does not know about. I think we have an issue with the storage model in this case. @rotorman I think the extra sources break the storage format in some way. I believe a model stored with `flyskyhallsticks` is different when read on a radio with a firmware that does not have that option. This is probably an issue for companion.

elecpower, 2021-09-19:
Even when 100% yaml still need to be able to map settings between eeprom and yaml radios.

elecpower, 2021-09-19:
If extra sources added, this has likely offset every source above so the ripple effect will show up in many lists.

raphaelcoeffic, 2021-09-19:
> If extra sources added, this has likely offset every source above so the ripple effect will show up in many lists/places.

Yes, but this is a "special compilation option" that is not available by default. We need confirmation, but I think it makes sense to declare it as "known issue" until we have a real solution to it. Otherwise, this will end up in a new binary format conversion. This is much too late in the game for introducing a new one.

pfeerick, 2021-09-19:
PR#761 being merged means the next nightly *should* work properly for you with Companion ;) Feel free to yell, scream and stomp if it doesn't :)

closed 2021-09-19 by pfeerick

jmxp69, 2021-09-19:
761 cured the translation between Companion and Radio *so far*. I need to do some deep inspection to make sure. 

On the radio, if I set the input source to MAX, then read the models in from Companion, Companion would show CYC2 per the picture.
On Companion, if I set the input source to MAX, then pushed the models/settings via Companion to the radio, the radio would show the odd JSx for input source.

If I changed JSx to a switch, the radio goes into Emergency Mode and after rebooting, the input source remained JSx.
If I changed JSx to MAX, no reboot, and the input source remained MAX until I read it by Companion and which point Companion would show CYC2 again.

jmxp69, 2021-09-19:
FYI This little gem pushed Widgets too. So when widgets were setup correctly on the radio a read/write cycle from Companion changed every widget by two spaces. Widget sources were changed forcing manual modification of widget source to -2 on every one.

raphaelcoeffic, 2021-09-20:
> FYI This little gem pushed Widgets too. So when widgets were setup correctly on the radio a read/write cycle from Companion changed every widget by two spaces. Widget sources were changed forcing manual modification of widget source to -2 on every one.

The issue is that making `EXT3` and `EXT4` available would shift all the indexes by 2, which is what you observed. And this why we had to take it back (for now).