EdgeTX/edgetx #759: EM related to input sources

pfeerick, 2021-09-19:
Found this one out at the field when trying to reconfigure a model... fun times! 🤪 

Go into model settings (new model or existing model), input, and then set your source via autodetect (i.e move a switch)... it will jumpt to the correct switch, and everything will seem fine, until you press ENT, touch away from it, or sometimes it will immediately EM... will segfault on `simu`, and EM on TX16S with yesterdays and todays nightly. Selecting the source *directly* from the list seems to work fine though. Only appears to effect inputs page, not mixes.

rburrow87, 2021-09-19:
You beat me, was just about to post this one!

pfeerick, 2021-09-19:
This also has some relationship with https://github.com/EdgeTX/edgetx/issues/758 ... looks like we all basically found aspects of this bug at the same time! :)

raphaelcoeffic, 2021-09-19:
@rburrow87 @pfeerick I believe the same issue exists with anything using a SourceChoice / SwitchChoice, which is basically on all Input/Mixer source / switch drop-downs.

pfeerick, 2021-09-19:
I did try input switch and mixer source and it didn't seem to misbehave. Either way, PR#760 fixes it for me. I know John will test this ASAP ;) so it'll should be in the next nightly...

PR #760 (2021-09-20): [color] do not use setValue() in SourceChoice / SwitchChoice
This potentially deletes the source/switch choice, which then triggers a "use-after-free".

Resolves #759
commit 64a87812 (2021-09-20): [color] do not use setValue() in SourceChoice / SwitchChoice

This potentially deletes the source/switch choice, which then triggers a "use-after-free".

Resolves #759
closed 2021-09-20 by pfeerick