EdgeTX/edgetx #762: Compile warning

robustini, 2021-09-19:
I've this compile warning in WSL, in the past I had many others but they have all been solved, this unfortunately remains:

/home/marco/edgetx/radio/src/opentx.cpp:758:4: warning: #warning "KEYSTUCK Message Not Yet Implemented" [-Wcpp]
   #warning "KEYSTUCK Message Not Yet Implemented"
    ^~~~~~~
cc1plus: warning: unrecognized command line option '-Wno-address-of-packed-member'

raphaelcoeffic, 2021-09-19:
> I've this compile warning in WSL, in the past I had many others but they have all been solved, this unfortunately remains:
> 
> [ 68%] Building CXX object radio/src/CMakeFiles/firmware.dir/opentx.cpp.o
> /home/marco/edgetx/radio/src/opentx.cpp:758:4: warning: #warning "KEYSTUCK Message Not Yet Implemented" [-Wcpp]
> #warning "KEYSTUCK Message Not Yet Implemented"
> ^~~~~~~
> cc1plus: warning: unrecognized command line option '-Wno-address-of-packed-member'

@robustini Hey! Nice to see you here! I guess we will just remove that warning, as I don't really see this as being a good thing to implement. But maybe it's just me.

pfeerick, 2021-09-19:
What, you don't want the warning that you have a key stuck on the transmitter? :-P

pfeerick, 2021-09-19:
@robustini `"#warning "KEYSTUCK Message Not Yet Implemented"` is one of many specific messages that were added to remind that a feature has not been implemented on a specific target... i.e. if you use a B&W radio, you won't see it, as they have the implementation of the "key is stuck" warning... but colourlcds don't, hence why it gets shown if you are building on a colour lcd. It is nothing to worry about, and is perfectly normal.

robustini, 2021-09-19:
Damn, during model conversion I lost all the pages I created in TELEM! :-P

raphaelcoeffic, 2021-09-19:
> Damn, during model conversion I lost all the pages I created in TELEM! :-P
> I was full in all over thirty models.

Obviously someone did not read the release notes / FAQ 😇

robustini, 2021-09-19:
You know how I am, first I try then I read! :P

closed 2021-09-19 by robustini

pfeerick, 2021-09-19:
Welp, well at least that's what backups are for :)

robustini, 2021-09-19:
> 
> 
> Welp, well at least that's what backups are for :) Silly github... no sad face "reaction"

I believe I have at least four backups in different places for all of my three TXs.
It remains a hassle to have to copy all the TELEM sections, I will do it when there is a fairly debugged version of Companion.

raphaelcoeffic, 2021-09-19:
> I believe I have at least four backups in different places for all of my three TXs.
> It remains a hassle to have to copy all the TELEM sections, I will do it when there is a fairly debugged version [...]

You should really drop by at our discord chat and discuss things with us ;-) I have a couple ideas to get you covered which we could implement in the next releases.