EdgeTX/edgetx #766: TX16S FLYSKY_HALL_STICKS conflicts with Trainer in Slave mode

rotorman, 2021-09-19:
If current main branch firmware is built for TX16S with FLYSKY_HALL_STICKS option, then trainer cannot be set to slave mode - this disables the FlySky digital sticks due to the usage of same DMA1 Stream2 channel.

I created a page in dev. Wiki that lists the DMA mapping on TX16S, where the conflict becomes also easily visible:
https://github.com/EdgeTX/edgetx/wiki/DMA-mapping-of-TX16S

raphaelcoeffic, 2021-09-19:
Other solution, IMHO more preferable: re-write Trainer slave output to use only IRQ. As the signal is really slow, it should be more suitable.

richardclli, 2021-09-24:
> 
> 
> Other solution, IMHO more preferable: re-write Trainer slave output to use only IRQ. As the signal is really slow, it should be more suitable.

Yes, agreed, always leave DMA to faster I/O

commit 453e478d (2021-11-27): Ported trainer output to IRQ driven mode

Next period is loaded after CCR2 expires (PPM delay).

Resolves #766
closed 2021-11-27 by raphaelcoeffic