EdgeTX/edgetx #767: Scrolling using touchscreen in model settings causes emergency mode

dannybgoode, 2021-09-20:
Tx - Radiomaster TX16S
EdgeTX v2.40 
SD Card SanDisk Extreme Pro 32gb

I have had two models now whereby using the touchscreen to scroll through the initial model settings screen (where you set up the timers, bind etc) causes the Tx to drop into emergency mode.

On one model I noticed the Timer 2 time value was corrupted and just showed a random ASCII character however the more recent one looks ok.  Apologies, I have deleted and reset the model with the corrupt timer value.

Both models are simple fixed wing planes.

Scrolling very slowly using the touchscreen can sometimes avert this and using the scroll wheel seems to avoid this happening all together.

I am more than happy to share model files etc, just let me know if you need anything.

pfeerick, 2021-09-20:
Have you tried a recent nightly build yet? There was a timer conversion issue fixed about a month ago, and I've not had any problems recently that seemed to manifest when scrolling, but it is always possible that it is model data related.

dannybgoode, 2021-09-20:
Thanks for the reply - I have not as I have tried to avoid nightly builds for stability reasons.  I will try the latest one though and see if the issue is still there.

I have other models set up with the same timers though and they work just fine.

nightflasher, 2021-09-21:
Okay, same happened to my NV14 - but edgeTX 2.5.0 RC1.

raphaelcoeffic, 2021-09-21:
> Okay, same happened to my NV14 - but edgeTX 2.5.0 RC1.

Could you please upload your RADIO and MODELS directory here? (Zip file)

dannybgoode, 2021-09-21:
My RADIO and MODELS directories as per the Zip.  Note it is the Spacewalker model that is failing and before that it was the Acrowot one but this seems OK at the moment.

[20210921 EdgeTX.zip](https://github.com/EdgeTX/edgetx/files/7203604/20210921.EdgeTX.zip)

pfeerick, 2021-09-22:
I had a quick flick through all the models in the "Planes" group/category, and it seems to be behaving for me... quick and slow scroll/slide with the touchscreen, wandering through various tabs and settings. I am running 2.5.0-RC1  for what its worth, but nothing was really touch in the last few days/weeks that should have changed this.

dannybgoode, 2021-09-22:
Hmm, that makes things a bit awkward! It was fine on that model for a couple of weeks and then just started happening.

same with the other one where the timer value became corrupted. I will try 2.5.0 just to be sure but if that doesn’t work I can’t risk flying on Edge :(

raphaelcoeffic, 2021-09-22:
> Hmm, that makes things a bit awkward! It was fine on that model for a couple of weeks and then just started happening.

That hints at something model specific.

> same with the other one where the timer value became corrupted. I will try 2.5.0 just to be sure but if that doesn’t work I can’t risk flying on Edge :(

Please try, so we at least know if this is still an issue or not.

dannybgoode, 2021-09-22:
I will happily try and see if 2.5 cures it and am happy to do any other testing that may help also.  I will let you know...

dannybgoode, 2021-09-22:
I have tried with RC 2.5.0 and the issue is the same.  I go into Model Settings and try and scroll down the first screen by 'flicking' up the touchscreen and it gets as far as the timer 2 settings and goes into emergency mode.  Scrolling slowly and it seems fine.

Now, interestingly I have noticed that the persistent setting showed corrupted characters that I am pretty sure were not there before.  Clicking on the setting allowed me to revert to 'On'.  I will post a screenshot...

dannybgoode, 2021-09-22:
[image]

dannybgoode, 2021-09-22:
Screenshot of corrupted text.  I have the model directory saved from yesterday so will copy it back across and see if it is corrupted there

[image]

raphaelcoeffic, 2021-09-22:
> Screenshot of corrupted text. I have the model directory saved from yesterday so will copy it back across and see if it is corrupted there
> 
> [image]

Thx! The corrupted text is most probably related to the issue!

pfeerick, 2021-09-22:
> Screenshot of corrupted text. I have the model directory saved from yesterday so will copy it back across and see if it is corrupted there

Also, if that model (and version of the model config) is in the zip file, can you tell me which one it is :)

dannybgoode, 2021-09-22:
Now, perhaps interesting.  I have copied the model back across and the Persistent value for timer 2 is not set at all however this is not an option in the menus (ie you can only choose Off, Flight or Manual Reset)

I wonder if there is an issue when transferring models from OpenTX where this value is not being recognised properly?

How do I find which model number it is? I have 39 models in my directory and only those created in Edge show the model name in the script, the others all just start OTX8

[image]

pfeerick, 2021-09-22:
Category and model name should be fine?

dannybgoode, 2021-09-22:
Category is Planes, model is Spacewalker :)

pfeerick, 2021-09-22:
Ah ha! Got the bugger this time... It isn't EM'ing, but I'm getting the garbage for Timer 2->Persistent :)

dannybgoode, 2021-09-22:
I wonder then if when fast scrolling the 'garbage' it is causing an intermittent error and throwing it into EM?  This morning the error was intermittent but was happening more times than not.

If the garbage is not a bug with Edge then apologies but given that people seem to be put off the moment their Tx goes into EM just once I thought it worth flagging :)

pfeerick, 2021-09-22:
Anything that triggers EM is worth reporting... as one way or another we have to try and prevent it... as we really only want EM to only happen when something more catastrophic like SD glitch or something happens... 

The question now is 1) how did the config get that way (i.e. if it is a conversion glitch, that needs fixing) and 2) how to make the code more robust to that. I see on companion it doesn't know what to do with that also, so just shows it blank (even though, again, that's not actually an option).

dannybgoode, 2021-09-22:
Let me know if I can do anything to help.  I still have the RADIO and MODELS backup from OTX from just before I moved to Edge if they would help at all.

I think it is to do with Timer 2 persistent value as well as the other model that failed was a copy of the one that we have just been looking at although I copied that in EdgeTX so it pulled the NULL value across with it.

I guess this is the fear for me.  If it is an issue, if it happens on someone else's Tx and it is a template model they copy many times it could cause significant issues.

pfeerick, 2021-09-22:
Yes please :) Then we'll  know for certain if it's a conversion bug that needs squashing ;)

I do remember there being a bug in timer conversion that was fixed after 2.4.0, but I think it only affected Throttle Start/ON being swapped on importing an OTX 2.3 config.

dannybgoode, 2021-09-22:
As per the Zip below.  I went straight onto the full release version of 2.4.0 if that is relevant.

Let me know if you need anything else from me.  I am no coder however I can find my way around pretty complex stuff with a small amount of guidance...

[20210809.zip](https://github.com/EdgeTX/edgetx/files/7209878/20210809.zip)

raphaelcoeffic, 2021-09-22:
@pfeerick I believe that could be the cause: https://github.com/EdgeTX/libopenui/commit/b21dedd73262f2c0d2577f94edcd80ffc6a01933

pfeerick, 2021-09-22:
Ok, we fixed that problem ... and caused another... so progress? 😁 Basically the UI code didn't sanity check values before processing them, so now it does... but I think perhaps a bit too thoroughly! 😆 So expect a firmware build within the next day that you can test to see if this is fixed :)

dannybgoode, 2021-09-22:
Thank you for looking at this so promptly.  Let me know when the build is ready and I will happily test it out.

Apologies for straying slightly OT and just for my understanding; what is the purpose of Emergency Mode?  Is it just a safeguard to prevent the Tx from totally bricking in the event of something unexpected happening or is there more to it?

raphaelcoeffic, 2021-09-22:
> Thank you for looking at this so promptly. Let me know when the build is ready and I will happily test it out.
>
> Apologies for straying slightly OT and just for my understanding; what is the purpose [...]

It’s the same as the windows safe mode. It has only the bare minimum to run.

dannybgoode, 2021-09-22:
Ah OK - I just wondered what the extent of the Tx's capabilities were in EM as it would be useful if it could maintain RF out so you could at least land your model as it seems to me then the Tx should just shut down.  That said I have no idea of the complexity of that thought so appreciate it may be impossible and there may be reasons behind the scenes to keep the Tx powered up :)

pfeerick, 2021-09-22:
Yes, the that is the whole reason for EM - it basically kills all the UI tasks, Lua scripts, etc, and just focuses on trying to keep the RF and inputs working ;)

dannybgoode, 2021-09-23:
Ah, in that case EM on mine is doing something different as I lose RF and the Rx goes into failsafe...

pfeerick, 2021-09-23:
Do you only lose it for a couple of seconds, or does it stay in failsafe?

dannybgoode, 2021-09-23:
When I was testing at the field it stayed in failsafe until I restarted the Tx and then picked back up.  Happy to test further if you can give me a way of forcing EM mode to be sure.

pfeerick, 2021-09-23:
No, that's fine... it's just not acting like I expected, so either I got it wrong, or perhaps there are certain configuration combinations it can't restart from...

pfeerick, 2021-09-23:
Right, so it looks like finally squashed that insidious 'lil bug that was playing havoc... if you would be so kind as to give this firmware a spin, and see if it fixes the garbage values and hopefully also 'nix that EM also. If out of range values are seen for any reason in choice dropdowns, it should now ignore them. 

[tx16_2.5.0-rc1_choice-fix.bin.zip](https://github.com/EdgeTX/edgetx/files/7217531/tx16_2.5.0-rc1_choice-fix.bin.zip)

dannybgoode, 2021-09-23:
Lovely - thank you for looking at this.  I will roll my Tx back to OTX and then reinstall and re-import my old OTX models to give it a full work through from scratch...

dannybgoode, 2021-09-23:
Just for clarification - should EM allow control of the plane with this version or should it just prevent the Tx from dropping into it by stopping the corrupt characters?  Just so I know what I am testing :)

raphaelcoeffic, 2021-09-23:
> Just for clarification - should EM allow control of the plane with this version or should it just prevent the Tx from dropping into it by stopping the corrupt characters? Just so I know what I am testing :)

It should re-connect to the RX. However, as we have not tested it for a while, I‘d say it‘s probably broken right now.

dannybgoode, 2021-09-23:
Thank you.  If it drops into EM I will let you know what happens...

pfeerick, 2021-09-23:
> should it just prevent the Tx from dropping into it by stopping the corrupt characters?

^^^ This. It should stop it happening in the first place. The next order of business will be to find out why EM isn't behaving... i.e. maybe it isn't co-operating with the MPM... but that would be the focus of another issue.

dannybgoode, 2021-09-23:
Understood.  If it goes into EM or if there is a way to force EM then I am happy to test if and when needed...

pfeerick, 2021-09-23:
Funnily enough, I'm going to now have to figure out how to do exactly that - force EM on demand - so that I can start testing that out... but that's a tomorrow problem! 😆

dannybgoode, 2021-09-23:
No worries - just tag me if you need me to try anything at any point on that.  I have just flashed back to OTX with the models as they were just before moving to Edge so will test a bit later and let you know what happens :)

raphaelcoeffic, 2021-09-23:
> Funnily enough, I'm going to now have to figure out how to do exactly that - force EM on demand - so that I can start testing that out... but that's a tomorrow problem! 😆 I think the version tab is going to become buggy for my own builds! 😇

There is a trick 😇

pfeerick, 2021-09-23:
Well, don't leave me hanging @raphaelcoeffic !! Is it pulling the (micro)SD card out by any chance? :laughing:

closed 2021-09-24 by pfeerick

dannybgoode, 2021-09-30:
Sorry for the delay in reporting back.  So, all seems good.  I rolled my Tx back to OTX and reinstalled Edge using this firmware and the models transfered fine.  no dodgy values and I have been using the Tx extensively and it is fine.

One thing I have noticed is that I can copy across the models I backed up from Edge prior to noticing the issue so the timer value is still Nul (shows as ---) however despite me using the Tx plenty it has not caused any issues.  Ie these were models with the corrupt timer values from the last time I moved from OTX to edge however it seems OK anyway and doesn't put in corrupt values just the ---

I can change these to Off, Flight or Manual Reset without issue.

If you need any more feedback from me let me know :)

pfeerick, 2021-09-30:
No problem, life tends to get in the way, and you did need to properly test it ;)

Thanks for the confirmation - so it seems like there is a conversion glitch somewhere, but sanity checking of the input before just  it and crashing is working properly ;) At least this won't be a problem after 2.5 is out of the way :)

commit ecbe89c0 (2021-09-24): Update libopenui - Fixes #767
commit 081c5dfa (2021-09-24): Update libopenui - Fixes #767