EdgeTX/edgetx #772: Change Model Select process so less likley to accidently select Create Model

aquaaddict, 2021-09-21:
Given I am selecting a model, and I am presented with the popup combo to select from Select Model, Create Model, Duplicate Model, Delete Model, it is very easy to accidently select Create Model instead of Select model when using the touch screen (espically outside in the bright sun with reduced screen visibility), as it is adjacent to Select Model, that then creates a new model and selects it as active.

Possible ways to overcome this accidental option selection may include:
- Enlarge the Select Model entry easier to select by touch without accidently touching neighbouring options (appreciate this could be difficult to do using standard combo list and would have to change combo to support some list item being displayed differently or use some other method to present the options to select from). 
- Add a confirmation to Create Model (Are you sure you want to create a new model? Create / Cancel)
- Instead of listing the Select Model and Create / Copy / Delete options all on the first popup, move the latter to a 'Manage Model' sub menu, the the combo then contains just Select Model , Manage Model, and if the user selects Manage Model, then sees the  Create / Copy / Delete option.
-

pfeerick, 2021-09-21:
Maybe a separator line space between Select and the other options?

aquaaddict, 2021-09-21:
yes that could be a good alternative, I am guessing it is the simplist solution to implement if you can add it to the combo as part of the list that gets passed in

Mantamatt1, 2021-09-23:
Also I have a habit of selecting an already selected model and cos of that the select model option is replaced with create model and what do I do... create another model. Maybe if you select a model your already on the first option could be cancel?

rhmaddox, 2022-11-07:
I like tie idea of a manage model option.  I tend to accidently hit the **duplicate model** and get no confirmation.  At lease at Are you sure to the duplicate model option.

raphaelcoeffic, 2022-11-08:
Normally you would ask for confirmation on destructive operations (like delete).

That being said, we could somehow find a way to better separate these.

pfeerick, 2022-11-08:
Delete does, but duplicate should have a prompt also. And I want to see if we can change the underlying new model workflow slightly also in 2.9 - at the moment it will create a new model, and *then* ask what to do (blank model, template, wizard) ... I'd prefer it ask, and *then* create the needed blank model... so if you hit create model, and then back... nothing happened. i.e. if you don't chose blank model, template, wizard, a model won't actually get created. Other than than, it won't be any different to the end user (other than a model didn't get created and switched to by mistake).

gagarinlg, 2023-03-18:
This is fixed in main

closed 2023-03-18 by gagarinlg