EdgeTX/edgetx #773: [LUA] Screen flickering in stand-alone mode

JimB40, 2021-09-21:
@jfrickmann for last several night builds when running script in stand-alone mode (Execute or TOOLS) screen is flickering horribly when redrawing, scrolling etc. Thought at first it is about graphic functions used in Apps like TBS AL but even simple text LUA behaves like that. 
https://discord.com/channels/596350022191415318/733424557636976692/888188330397798451
 
It was okay in 2.4 and later night build so seems it's about LUA parser implementation changes

pfeerick, 2021-09-21:
Also mentioned in #635 ;) So it goes back close to a month at least ;)

jfrickmann, 2021-09-21:
So the discord link posted above does not work for me. But looking at the Youtubes on #635 -

I would say that the first one must be related to the display buffer implementation, and that is waay over my paygrade. And TBH, I do not consider a short flicker when a new app fires up to be a big issue. 

But I agree that the second one is pretty annoying. Does it only happen with the TBS script, or also with other scripts?

Can we pinpoint the commit or PR where the issue started? I have not been tinkering with any of that stuff since July 22nd when pre-emption was added. The only other commit is for colors on that file ([standalone_lua.cpp](https://github.com/EdgeTX/edgetx/commits/main/radio/src/gui/colorlcd/standalone_lua.cpp))

I remember that you had an issue back in the beginning with event handling, where your radio did strange things. Did it just go away by itself, or did you do something to solve it?!?

JimB40, 2021-09-21:
Thing is that to exclude too many factors I'm taking into consideration just TBS AL that is not changed for more or less 2,5-3 months. Same LUA binary.
So with 2.4 it was working okay no flickering. then 3-4  weeks ago started to flicker.
I will make some more tests but definitely something changed in a C++ code.

pfeerick, 2021-09-22:
I believe the short flicker / display buffer issue can be remedied by clearing the display as early as possible... I had that problem from *before* 2.4.0 was released when playing with some Lua code, and it was a multi-stage script at that,  and it was just a matter of clearing the screen in the first stage, not at a later stage ... 

I can't comment on the flickering...

raphaelcoeffic, 2021-09-22:
@jfrickmann could it be related to the way the draw routine is preempted? If the script is preempted before the drawing is done, what happens? At the moment, LUA does not use double buffering, so that it must produce an image of the UI thread tells it needs one. If however, the script is preempted before it’s done, we would still copy the buffer (no choice), so that such an effect would be seen.

jfrickmann, 2021-09-22:
I would not think so. First of all, I tested preemption and back then it worked. Besides, I do not assume that the script actually does preempt.

Robert had some other strange problems with touch back when he first tested that. I never heard the explanation but that part seems to be working now. I was wondering if the TBS scripts he has running in the background have something to do with it?

I have not yet had time to load 2.5rc1 on my Tx to test, but has anyone else experienced the flickering issue?

At this point I do not have any suggestions for why it is happening.
————————
Jesper Frickmann

JimB40, 2021-09-23:
Guys if you touch subject, there is also one more thing.
Once script is executed in stand-alone mode there kind of 'last used app screenshot' during initialization.
I mean last app screen is visible until first lcd command is executed in new app.
If app has long initialization procedure it will be visible even for 2-3 seconds (from the moment you press Execute to the first command like lcd.clear)
For me it looks like display buffer is not cleared.

raphaelcoeffic, 2021-10-19:
According to tests from @JimB40 & @richardclli, it seems to be linked to this commit: 8e4e2cceeaa1c2ab901e962b561b11c6829f46b3

@jfrickmann does that help?

jfrickmann, 2021-10-20:
Not really. What I need is a script that can reproduce the problem, so I can start tracing to see what happens.

@JimB40 , do you have a Lua script that I can use to reproduce the problem, so I can start troubleshooting?

JimB40, 2021-10-31:
Well it seems this flickering happens with MCU intensive scripts. 
Easy  lcd.drawText(0,0,'Hello world',0) won't trigger it. I understand source code would help but with AL is not possible.
btw Discord or Adobe don't disclose source code to LInux dev team. Do they? :)

My suspicions goes
1. preemptive way of working. From time time "attempt to yield across C-call boundary" LUA error shows. Like with table.sort() on table with lots of entries. Haven't time to read about it.
2. Key events handling as it happens with fast scrolling. 

No chance to write so big app just to test but maybe I'll play with simple dummy MCU heavy routines to check.

raphaelcoeffic, 2021-10-31:
@JimB40 reproducing it with something that uses similar patterns is a good idea. Talking about Linux: if you want a bug fixed in Linux kernel, you’d better have some source code, other wise they will send you to hell ;-)

jfrickmann, 2021-10-31:
> ... btw Discord or Adobe don't disclose source code to Linux dev team. Do they? :)

Do you think that Adobe would post an issue on kernel.org like this: "Hey, the latest kernel is causing this issue with Adobe Acrobat. You cannot see the source code, and we will not tell you what the program does. But please fix it ASAP - thanks!" ?

Companies like Adobe likely have their own team of Linux developers, and if there is an issue, they will look into it. If they find that the issue is related to the Linux kernel, then they will submit a patch to kernel.org. And that's the power of open source - everyone can contribute! So you are welcome to do the same. Get setup for building EdgetTX, spend as many hours studying the code as I have, and submit a pull request when you have identified the issue and made a fix that does not break anything else.

But if you do not want to do that, and you want my help, then you must help me help you. Otherwise, I cannot help you. I want to help you, but as long as you do not work with me, I am not able to help you.

And on that note: did you test the PR that I asked you to test above? (https://github.com/EdgeTX/edgetx/pull/532/checks)

Sincerely,
Jesper

JimB40, 2021-11-01:
> And on that note: did you test the PR that I asked you to test above? (https://github.com/EdgeTX/edgetx/pull/532/checks)

Nope. Check expired
[image]

rotorman, 2021-11-01:
@JimB40 PR #532 has been merged to main branch, thus you can get it from the std. nightly:
https://github.com/EdgeTX/edgetx/actions/workflows/nightly.yml

jfrickmann, 2021-11-02:
Nevermind checking that commit, I don't think it is the issue. 

Since I do not have any of the information that I need to solve this issue, I will start a branch and make a few random guesses from the hypothesis that Lua preempts and there is a delay before it is called again. I will try to remove whatever I think may introduce such a delay.

jfrickmann, 2021-11-02:
@JimB40 please test #982 and let us know if it is better.

JimB40, 2021-11-02:
Okay so here is my finding after testing.
Much better. At first I thought it's not flickering at all. 
After some more extensive play I've noticed that "last displayed" elements flicker

I have very similiar GUI structure like @jfrickmann in LibGUI
There is table "Elements" that holds every displayed GUI element in order they should be displayed on screen
To give example"
Elements[1] = background
Elements[2] = logo
Elements[3] = window1
Elements[4] = window2
Elements[5] = window3

Then windows have their Elements table
window1.Elements[1] = button
window1.Elements[1] = editable _field
window1.Elements[1] = label
etc.

So in run() Elements table is iterated 0-#max and all sub-tables with windows elements building proper GUI.

In my test if Elements table has 2-3 entries nothing flickers. Then if Elements table has 4-5 entries and 3rd one is window with option list (more MCU processing) last GUI items start to flicker. ie Elements[5]
Like there was not enough time to draw them (and they are moved to kind of next cycle?)

JimB40, 2021-11-02:
One more finding. I thought it's connected to how much LUA script lines are executed in run()
Example:
Serial protocol parser is event based so once packet is detected it is parsed in the same cycle.
But there are App states like parameter editing when device is not pooled so parser is not invoked and this last elements flicker happens.
So my conclusion is that is solely connected to how many items are in Elements table to be displayed.

jfrickmann, 2021-11-02:
@JimB40 - great, it sounds like we are making progress! Since your GUI is very fancy looking, I believe it can hit the 50 ms time limit and preempt. And as you suggest, it probably waits a cycle before drawing the last elements.

It is inevitable with preemption that we reach this point as we keep adding more demanding graphics elements to the screen. The question, however, is if should we give standalone scripts some extra time before preemption? The tradeoff is, the longer time we give the scripts, the longer they can hold up everything else in the GUI main thread. Without preemption, there is no time limit but an instruction limit, and time/instruction can vary. Standalone scripts get 20000 instructions and other scripts get 10000 instructions per cycle in OpenTX.

What do you say @pfeerick and @raphaelcoeffic ? Is e.g. 100 or 200 ms a reasonable time for a standalone script to hold up the GUI thread?

jfrickmann, 2021-11-04:
I have been looking at the screen buffer implementation. I want to add some traces and see how it behaves when a script preempts. It _should_ only refresh the buffer when the script returns from the function call, but does it? I will get back when I have tested that. After this weekend, where we have our fall F5J contest.

pfeerick, 2021-11-04:
Good luck (on both fronts)! ;)

JimB40, 2021-11-04:
> What do you say @pfeerick and @raphaelcoeffic ? Is e.g. 100 or 200 ms a reasonable time for a standalone script to hold up the GUI thread?

All about finding sweet-spot. Maybe trial & error will work.

jfrickmann, 2021-11-05:
So I ran the trace with a standalone script that preempts, and it did exactly what it was supposed to do:
```
20.43s: ----> luaTask -
20.43s: ----> luaTask +
20.48s: ----> yielding
20.51s: ----> luaTask -
20.51s: ----> luaTask +
20.53s: ----> returned
20.53s: ----> invalidating
20.53s: ----> painting
20.57s: ----> luaTask -
20.57s: ----> luaTask +
20.62s: ----> yielding
20.65s: ----> luaTask -
20.65s: ----> luaTask +
20.67s: ----> returned
20.67s: ----> invalidating
20.67s: ----> painting
```
"luaTask -" is when Lua gets called in the background (nothing happens as standalone does not have background functions), "luaTask +" in the foreground, "yielding" = Lua waits until next cycle before it continues, "returned" = the Lua run function returned.

As you can see, the GUI politely waits until Lua has returned before invalidating the window and then calling paint to copy the Lua LCD buffer over.

@raphaelcoeffic - please confirm that I have understood this correctly: StandaloneLuaWindow::paint(...) copies the Lua LCD buffer over to the screen, right?

jfrickmann, 2021-11-06:
OK, I have found out why it sometimes flickers after a key was pressed. It is because the standalone GUI calls Lua twice per cycle, one in onEvent() and then again in checkEvents. If the Lua function returns from the first call and invalidates so the screen can get re-painted, and then preempts during the second call, then luaLcdBuffer will not be ready when paint() is called.

@JimB40 , please check #982 with the latest commit and let me know how it works.

JimB40, 2021-11-08:
Well seems like you pinned down this bug, Sir :)
Feel slightly slower but no sign of flickering

jfrickmann, 2021-11-08:
Great to hear! So I assume that Peter can close this issue then?

closed 2021-11-08 by rotorman

pfeerick, 2021-11-08:
FYI, convention is to *leave* an issue open until the fix for it is actually merged into codebase, as until then the fault is not actually *fixed*, just that there is a potential fix pending ;) Thus is part of the reason issues are linked to pull-requests, so this is managed automatically. Additionally, this means when someone comes along to report in between a PR being created, and it being merged, they see there there is already an open issue so won't report it again.