EdgeTX/edgetx #776: [LUA API] lcd.RGB() name change

JimB40, 2021-09-21:
Continues LUA API naming scheme.
lcd.RGB -> lcd.setColorFlag()
because this is what function actually does

To maintain legacy lcd.setColorFlag should be main function and lcd.RGB should point to it and will be depreciated over time.
@jfrickmann @raphaelcoeffic

jfrickmann, 2021-10-30:
We should not change the names of existing Lua functions for the following reasons:

1. What is the "right" name is a subjective opinion. If a name is really so bad that it needs to be changed, then it must be based on a previous community discussion resulting in general consensus for the new name.
2. Name changing does not add any new functionality or other improvements to the software.
3. The time spent changing the name of an API function is not only the time spent doing the change itself, but much more the time spent by script developers discovering that their script no longer works and then editing scripts.
4. Some members of the scripting community will be very upset and unhappy about that; especially in light of the fact that no new features or improvements were added.
5. Choosing a name is the prerogative of the person who implements a new function.

So please do not mess with the names of existing functions and do not merge #952 and #953!

pfeerick, 2021-10-30:
I agree with all of this in principle, particularly the first point, which is one reason I simply haven't merged the PRs. I will however point out that they are *not* changing the name, which is exactly what is stated/detailed in the last line of this issue ;) The goal is to standardise on names, and at a future date they can be depreciated with suitable advance warning, but only as part of a **major** version chance, in accordance with semver standards. i.e. ETX 3.0 would be the earliest time for that to happen, and I don't think that would happen so soon anyway.

jfrickmann, 2021-10-30:
OK. But I still think that we should focus on fixing bugs and adding features rather than changing names 😃