EdgeTX/edgetx #779: Proposal to Deal with Screen On Bright Days

jmxp69, 2021-09-21:
One of the biggest complaints I see about the screens on these radios is that the sun washes them out and they are difficult to see outside. I have to agree, it can be challenging. Today, there is a simple backlight brightness mechanism that can be defined in global functions. I also know JIMB40 is working on a theme tool. 

Imagine This:
• Backlight Global Function Defined to work from S1.
• At 90% of S1, the backlight is set to 100%
• Above 90%, the screen theme automatically switches to a high contrast color scheme (theme tool) to make visibility outdoors much easier.

I think of this in use during flight when I might want to glance down to get a look at telemetry values, timers, output monitor etc. Not while actually "using" the radio to configure or setup models. What can we do with the hardware and software on the radio right now to deal with bright sunny days outside?

pfeerick, 2021-09-21:
Hm... I can envisage this is almost possible already... but would be model specific because of logical switches being model specific rather than global... i.e. have your global S1 brightness control, and a logical switch that monitors the S1 as well, and goes true when it's at 80 or so. This is then the trigger for a special function runs a lua script... since lua scripts can alter all the theme colours now ;) 

If you want it entirely global as things stand right now, you could instead simply have a global function run a lua script in response to a switch (i.e. use those often T5/T6, or use 6 position buttons)... with said lua script being the script responsible for setting the theme colours... ;)

JimB40, 2021-09-21:
Like idea. but this is functional feature.
I'd rather extend LUA API for lcd.getBrightness() and go with small widget to make it happen

pfeerick, 2021-09-21:
lol... true... but that means you need to write the lua script / widget to do that! ;)

JimB40, 2021-09-21:
So? I will and include it in ETXSD card distro. Don't tell me compiling with every small functional change firmware is better :)

pfeerick, 2021-09-21:
No, just that I prefer leaving Lua scripting to those that know Lua already! 😆 Plus offering up an alternative for those that don't know Lua ;)

JimB40, 2021-09-21:
For those who don't know LUA UX will be the same just LUA code executed not C++ :)

MervDale, 2021-09-22:
I would think that if you are flying taking your eyes of your model is not a good idea. These radios have great voice data options have you thought about having the radio voice all or any of the data you think you may want while flying?  That's what I do!

pfeerick, 2021-09-22:
> For those who don't know LUA UX will be the same just LUA code executed not C++ :)

Actually, now that I think if it, since John talks about "telemetry values, timers, output monitor"... doesn't a certain FM2M Lua app do most of this, and with good contrasting colours? 😆 

> I would think that if you are flying taking your eyes of your model is not a good idea. 

That is a good point, but is not a rule. If I'm flying a docile Skysurfer or Bixler, which will basically fly on it's own (and it prefer the idiot behind the sticks to just leave it alone at let it fly)... looking away for up to 20 seconds isn't an issue at all. And then there is craft that have autopilots in them. I certainly agree with using the voice prompts where-ever possible... i.e. I have the timers voice prompt every minute, and countdown. Periodic voice prompts for battery voltage, as well as more insistent ones once the low voltage thresholds are reached. Primarily because at the present looking at the transmitter isn't an option, since I'm mostly flying FPV.

jmxp69, 2021-09-22:
As I've considered this further, I may have come up with even more ease of use:

As the knob is pushed past 90%, the main page could be set to the LAST page which the user could setup with the high contrast items of interest.

A global function could be defined that says something like: High Contrast Enable
Pot > 90% switch to last page
Brightness full
Apply high contract scheme
Pot < 90% switch to last used page
Brightness per system setting
Apply normal scheme

On the page setup enable High Contrast Checkbox which effectively means black and white or whatever combo works best. Then the page can be setup utilizing values/information/widgets user deems important.

jmxp69, 2021-09-22:
> I would think that if you are flying taking your eyes of your model is not a good idea. These radios have great voice data options have you thought about having the radio voice all or any of the data you think you may want while flying? That's what I do!
> Just to make a point of not taking your eyes of a flying plane, under the right sky conditions it does not take much to lose sight of a plane especially if it is edge on to your line of sight.

What if I'm not flying? What if my plane is on the ground? What if I just want to glance down at my voltage? What if I want to see where my trim switch is during flight? I'm well aware of the capabilities of the radio. I fully understand how and when to use voice. I don't need suggestions on how to fly. I know how to fly and have plenty of experience. I would think the correct thing to do is participate in answering the question asked rather to impose your governance on the person asking. And if you can't offer help on the question asked then consider being a spectator.

rburrow87, 2021-09-22:
Yeah the only time I'm looking at my screen _while flying_ is when it feels like an alarm should have gone off already and I want to quickly glance at the timer, voltage, or current consumption to see if I missed it. I fly at a club and some of the fellow pilots have louder aircraft, so it happens. Obviously if you feel unsafe looking at your screen in some situation, do not do it.

Anything to make it easy to glance at critical information is a welcome feature IMO. I like the idea of having some kind of high contrast mode that's easy to switch in and out of like this.

MervDale, 2021-09-22:
I offered an alternate solution besides the need to look at your radio screen while flying in bright sun light as glare and reflections can make it extremely difficult and as OTX and ETX both have excellent voice systems that tell you anything that would be of interest or critical while flying including snafu! Unfortunately I don't have a solution for manners!

pfeerick, 2021-09-22:
> Unfortunately I don't have a solution for manners!

And I don't think any is needed... I think that remark was a bit uncalled for, as not everyone will agree with a idea/proposal, so the person offering should be prepared for pushback, especially when their level of knowledge and understanding is often only known to themselves. I happen to know John is quite competent, but how were you to know? ;)

I'm starting to warm up to JimB40s idea more and more, as it sounds like the job for a Lua script, rather than having to have the radio firmware do the job. Then you can set your specific sensors you want shown, and have the colours set to high contrast, etc. And just have that on one of the pages of your setup. The beauty is of it being a Lua script is that it would also be available to all models, rather than a model specific panel setup.

JimB40, 2021-09-23:
I will be little stubborn :)
OS should provide framework to implement such a specialised functionality in LUA. 
Once is tested, adopted or required by larger amount of users it may be incorporated as standard OS feature.
In fact if extend LUA API with lcd.getBrightness I will write 3 lines widget for @jmxp69 in 5 mins :)

jmxp69, 2021-09-23:
> I offered an alternate solution besides the need to look at your radio screen while flying in bright sun light as glare and reflections can make it extremely difficult and as OTX and ETX both have excellent voice systems that tell you anything that would be of interest or critical while flying including snafu! If you are on the ground then you can always turn your back on the sunlight and use your own shade to read your screen. Unfortunately I don't have a solution for one's outlook!

You didn't offer a solution. You offered an off topic response suggesting how you feel I should fly my plane and I didn't ask for that advice. Your response is the reason I don't use forums. I didn't ask for suggestions on how to fly or use a radio.

jmxp69, 2021-09-23:
> I will be little stubborn :)
> OS should provide framework to implement such a specialised functionality in LUA.
> Once is tested, adopted or required by larger amount of users it may be incorporated as standard OS feature.
> In fact if we extend LUA API with lcd.getBrightness I will write 3 lines widget for @jmxp69 in 5 mins :)

Thanks Robert. Regarding implementation, I'm no expert so I don't have much of an opinion on how it's done. If LUA is the correct path then I'm right behind you. I do think this would be a valuable feature addition.

richardclli, 2021-09-23:
The best solution is to change a brighter LCD, I found a nice IPS LCD with 1000 NITS, which is very good even in bright sunlight.

JimB40, 2021-09-23:
@jmxp69 ETX is many ways similar to many other OSes we use on laptop, phone etc. There is set of core needed features included in OS and then good OS API to create functional apps. Nobody is considering to include Facebook functionality in iOS or Linux even user  base is quite huge :) Primary areas for OS is low level support for display, UI, sound, io etc plus in case of ETX stable and reliable RC link implementation.
Benefits of this approach is clear for me:
- easier code maintenance
- focusing on new radio support
- new drivers to cover upcoming MCUs
- more time to provide consistent and well documented OS API for App devs
etc, etc
We are growing but still small team. So I'm stubborn with stealing time from top level C++ coders for functional features that can be done "externally".  Not saying that throwing  good idea means stealing :)
Idea is definitely worth implementing but not on OS level.
:)

jmxp69, 2021-09-23:
@JimB40 High contrast and accessibility features (even light/dark themes) are core to pretty much most mainstream OS's these days. Accessibility has been a core feature of Windows since 3.0. which effectively enabled a high contrast mode on the primary display. I wouldn't view accessibility as an add-on application like Facebook.

pfeerick, 2021-09-23:
Agreed. High contrast and accessibility are indeed core features to an OS... but there are also different *levels* to an OS. i.e. it's not the job of the kernel to manage this sort of stuff... the kernel is what is responsible for managing all the hardware. The userspace is what is responsible for that sort of stuff, and this is where Lua scripts can come in... the radio firmware will provide a certain amount of basic functionally, but really should only be providing the *core* features  and providing all the hardware abstraction so that Lua scripts do the rest, and don't need to care about what hardware they are on... just that when they query for something like screen brightness, they get it. And when they call for something to be drawn to the screen, it's done. 

A Lua script can be used to either 1) stitch all of this functionality together or 2) provide a much cleaner alternative.

jmxp69, 2021-09-23:
Yeah honestly I really don’t care how it’s implemented. My only legitimate concern would be that it is connected to hardware so I wouldn’t have to press system, files, tools, brightness, execute. If it can be linked to the knob, slider, switch or bonus: logical switch then honestly I could care less how it’s coded.

MervDale, 2021-09-24:
I have removed my suggestions seeing as you are a such a patreon "expert" but I will say this when answering or offering a possible alternate solution here on GH and Discord and when trying help the Dev's get ETX  up and running one has no idea what experience the person has with either RC models or radios and considering that these hard working Dev's are doing all this in their own time and most likely costs. Offering some alternate options and answering the same questions is the least one can do rather than adding to all the other  I want, when I want it issues!

jmxp69, 2021-09-24:
What does patreon have to do with any of this? Again you show your penchant for going off topic. You need to learn how to either answer a question or watch. You made assumptions and this is the result. The proposal has to do with radio usability not coaching on how to fly.

JimB40, 2021-09-25:
Guys if I may ask not to stir up "war" in GH issue. It's very time consuming for us afterwards to filter out the, what is exactly about. :)

closed 2021-09-25 by jmxp69