EdgeTX/edgetx #780: [LUA API] ne function tele.isOn()

JimB40, 2021-09-21:
this refers #777

In current implementation to check if Telemetry data is coming we have to:
1. getRSSI() and check against 0.
2. getValue('sensorName') against 0. 

There should be simple function **tele.isOn()** returning true/false if telemetry stream is received to simplify and unify this checking if telemetry data is actually received

Caveat:
With newest protocol implementation (Ghost, CRSF, ELRS) there are some sensors that are actually send in Tx rather then Rx  I call it 'off-line' telemetry. Example in Ghost protocol all VTX sensors are stored in Tx and available (sending data) even Rx is not powered.

pfeerick, 2021-09-21:
Hm... is that as in some telemetry is actually from the module, not the receiver?

JimB40, 2021-09-21:
Yes it is :)