EdgeTX/edgetx #786: Switch, pot and throttle warnings should not disabled SF or LS

pfeerick, 2021-09-21:
**Is your feature request related to a problem? Please describe.**
Switch, pot and throttle warnings currently disable execution of special functions and logical switches before radio control is commenced, meaning they can't be used to detect problematic configurations in a more dynamic way. 

**Describe the solution you'd like**
Allow logical switches and special functions to run before radio control is enabled, allowing for more programable warnings to be issued.

**Additional context**
See more on on RCGroups where Miami Mike raised this idea - https://www.rcgroups.com/forums/newreply.php?do=newreply&p=47834747

eshifri, 2021-09-21:
Can't you just disable all warnings and use SF for validating anything and overwriting all controls?

raphaelcoeffic, 2021-09-21:
> Can't you just disable all warnings and use SF for validating anything and overwriting all controls?

As it seems, we cannot for now, as this disables running these at an early stage…. Go figure why! 😝

eshifri, 2021-09-21:
I mean user can disable all warnings and create SF to handle all switches and other controls. Won't it have the same effect?

pfeerick, 2021-09-21:
Not how I interpret the request. I think if you read the full post from RCGroups, I seem to think it would actually feed into the warnings system itself - i.e. a LS+SF combination could *programatically* trigger a waning, rather than it just be a generic switch warning. i.e. there is no problem starting with switch C in Acro position, but there is if you have throttle hold OFF and throttle UP. In other words, this is about making the warnings smarter ;)

eshifri, 2021-09-22:
That is my understanding as well, but can't user achieve the same without warning system?
SF plays track and calls some LUA script?
Looks easier to implement than call-back into the warning system.

pfeerick, 2021-09-22:
And how does SF block radio?

> Plus, the special aspect of the warnings screen is the RF is still turned off ;)

pfeerick, 2021-09-22:
Oh, indeed, if you just want to over-ride a specific channel once RF is on, that would be fine :) 

I feel the key aspect to the proposal is that LS+SF can trigger the warning screen and RF is still off, meaning it's a visual block, plus you get the audio prompt, plus anything extra that the SF triggers. And it can all be dismissed at the touch of a button if you need it to be... 

It require some refactoring of the warning screen  - i.e. a warning event would probably have to be added to special functions, perhaps with text input so a message could be shown on the warning screen, as well as logic to monitor for that event. But that's what this issue is for - to discuss if Miami Mike's idea is something that enough people would find useful to invest time on implementing it ;)