EdgeTX/edgetx #787: [NV14]  2.5.0 RC1 - not usable!

nightflasher, 2021-09-21:
I've flashed my "good old trusty" nirvana to edgeTX today and it instantly bricked it!
My models (originally stored on sdcard) where completely wiped, all radio data? Also wiped! As you all know, there was no companion to back up anything before - so I only had a three year old one.
After flashing edgeTX and starting the nirvana, some errors popped up, some of them where unreadable. After all the messages are gone, I've opened the model select menu - no models at all.
I've shut down the radio, switched on again and the radio only started with "EMERGENCY MODE".
I was able to flash the old firmware, restored my three years old backup and everything is fine now.

EdgeTX is not ready to be flashed on the nirvana as daily drivers.

raphaelcoeffic, 2021-09-21:
> I've flashed my "good old trusty" nirvana to edgeTX today and it instantly bricked it!
> My models (originally stored on sdcard) where completely wiped, all radio data? Also wiped! As you all know, there was no companion to back up anything before - so I only had a three year old backup.
> After flashing edgeTX and starting the nirvana, some errors popped up, which some of them where unreadable. After all the messages are gone, I've opened the model select menu - no models at all.
> I've shut down the radio, switched on again and the radio only started with "EMERGENCY MODE".
> I was able to flash the old firmware, restored my three years old backup and everything is fine now.
> 
> EdgeTX is not ready to be flashed on the nirvana as daily drivers.

You mean the models coming from the old version shipped by FlySky? If this is the case, this is normal. There is no support for this old format.

raphaelcoeffic, 2021-09-21:
By the way, you can backup everything by just copying all the files on SD card to a safe location.

nightflasher, 2021-09-21:
> 
> You mean the models coming from the old version shipped by FlySky? If this is the case, this is normal. There is no support for this old format.

No! I have the latest qba667/Stefan73 firmware from early this year (February/March, I think). Never had any issues upgrading the firmware made by them.

raphaelcoeffic, 2021-09-21:
> > You mean the models coming from the old version shipped by FlySky? If this is the case, this is normal. There is no support for this old format.
> 
> No! I have the latest qba667/Stefan73 firmware from early this year (February/March, I think). Never had any issues upgrading the firmware made by them.

Which is a very old format. You need to understand that this comes from an old OpenTx fork with a very special format.

raphaelcoeffic, 2021-09-21:
This is a duplicate of #516

closed 2021-09-21 by raphaelcoeffic

yds, 2021-09-21:
> I have the latest qba667/Stefan73 firmware from early this year (February/March, I think).
> Never had any issues upgrading the firmware made by them.

TL;DR: there is no model conversion from FlySky-OpenTX to EdgeTX on Nirvana NV14

@nightflasher EdgeTX does not convert models from qba667/Stefan73 builds -- I took pictures of every screen of my model before upgrading and redid my model from scratch after flashing EdgeTX to my Nirvana. best to use a fresh SD card prepped with EdgeTX-Flasher tool and leave your FlySky-OpenTX SD card alone so you can easily go back should you need to for whatever reason.