EdgeTX/edgetx #789: Add auto power-off option for radios with soft power control

pfeerick, 2021-09-22:
It looks like this is going to go stale on OTX so moving my bat and ball over here ;) 

I would envisage this as another option in the radio setup for "auto power off", with the ability to have a SF allowing it to be blocked, if there is a specific model that needs this to be prevented in some manner. The auto power off timer would be similar to the inactivity alarm, meaning stick moment, button press would reset it. It would also bee good to have an option for voice prompt/beep/haptic countdown say every 10 seconds in the last minute  - "Transmitter about to auto power down" or words to that effect. 

https://github.com/opentx/opentx/issues/6421

lshems, 2021-09-22:
Why not inverse. SF power off. Trigger on inactivity with delay. Inactivity as new general trigger.

Fits much better in the general concept.

pfeerick, 2021-09-22:
How is that inverse? Inverse is "opposite" ... and you're not proposing auto power on! 😝 

Yeah, that sounds ok I guess. I was trying to make it so it didn't just blindly link to inactivity... especially as it would need its own warning options (inactivity alarm != about to power off). But I can see the benefits of inactivity being a trigger anyway - i.e. you could indicate screen blanking, specific channel overrides, etc if wanted. Maybe that warrants a separate issue, related to this.

lshems, 2021-09-23:
The inverse was referring to enabling the power off by a special function instead of disabling it.

Enabling leading to execution with the delays that can be set on any function, so the the actual 'enable power off'would lead to the action 'power off' at some moment.

But yeah, points of view.