EdgeTX/edgetx #790: [LUA API] lcd.getTextSize (lcd.sizeText) covers only default font size

JimB40, 2021-09-22:
1. Implement other font sizes
2. Make note in luadoc it only covers default font size

jfrickmann, 2021-10-03:
What do you mean by "default size"? It works fine with the available font sizes, as far as I can tell.
```
local name = "Test"

local flags = {
  SMLSIZE,
  0,
  BOLD,
  MIDSIZE,
  DBLSIZE,
  XXLSIZE
}

local names = {
  "SMLSIZE",
  "NORMAL",
  "BOLD",
  "MIDSIZE",
  "DBLSIZE",
  "XXLSIZE"
}

local options = {
}

local function create(zone, options)
  local widget = {
    zone = zone,
    options = options
  }
  return widget
end

local function update(widget, options)
  widget.options = options
end

local function background(widget)
end

local function refresh(widget, event, touchState)
  local y = 0
  
  for i = 1, #flags do
    local w, h = lcd.sizeText("Egg", flags[i])
    lcd.drawText(0, y, names[i] .. ": Egg = (" .. w ..", " .. h ..")", flags[i])
    y = y + h
  end
end

return {
  name = name,
  options = options,
  create = create,
  update = update,
  refresh = refresh,
  background = background
}
```
[image]

pfeerick, 2021-10-03:
Perhaps it is that `getTextSize` only works for the default size?

jfrickmann, 2021-10-03:
There is not a function named `getTextSize` AFAIK...

pfeerick, 2021-10-03:
No, it's not... but I see "someone" was a bit presumptive :-P https://github.com/EdgeTX/edgetx/issues/774

philmoz, 2026-02-09:
Can be closed - lcd.sizeText function works correctly for all font sizes.