EdgeTX/edgetx #792: For NV14 enabling Internal module makes External module and Trainer unreachable

eshifri, 2021-09-22:
**Describe the bug**
If NV14 Internal Module is set to FLYSKY, there is no way to scroll the Model Setup screen to external module and Trainer setup without going to another screen first.

**To Reproduce**
On the model setup screen select Internal module FLYSKY.
Observe scroll bar set to the bottom of the screen and External module option is unavailable.
Go another TAB (e.g. Inputs) and return to the Model Setup TAB - External Module and Trainer will be present at the bottom of the screen.

**General information**
 - EdgeTX 2.5.0 
 - Hardware NV14

pfeerick, 2021-09-22:
btw, this isn't NV14 specific... it's related to the MPM section generally... as you've noticed if you go to another tab and back again, it's fine. You can make the same thing happen just by turning internal from off, to multi, to off again.

eshifri, 2021-09-22:
Yes, definitely the parent window size internal is not updated when these child windows are changed.
I do not see how to do it: the ModelSetup height is defined by the grid. The grid is filled when the window is built. Any changes to the internal windows do not effect the grid until we build this page again. Can we call build() for a parent every time the child is updated?

commit a4af85ef (2021-09-25): Fix module and trainer subform resizing and show NV14 RF module version (#800)

* Fixes #792
* Add RF module FW version
* Clean(er) implementation
* Whitespace changes cleanup
* Cleanup
* Trainer module fix
commit f8a0a445 (2021-09-25): Fix module and trainer subform resizing and show NV14 RF module version (#800)

* Fixes #792
* Add RF module FW version
* Clean(er) implementation
* Whitespace changes cleanup
* Cleanup
* Trainer module fix
commit 42a152e4 (2021-09-26): Fix module and trainer subform resizing and show NV14 RF module version (#800)

* Fixes #792
* Add RF module FW version
* Clean(er) implementation
* Whitespace changes cleanup
* Cleanup
* Trainer module fix
closed 2021-09-25 by pfeerick