EdgeTX/edgetx #794: QX7 & X9D+2019 - reset/bootloop

pfeerick, 2021-09-22:
For some yet unknown reason, both my X9D+2019 and also @JyeSmith QX7 (as well as another QX7 owner) have suffered reset or bootloops which seem to require a reflash (or perhaps just EEPROM erase/re-write?) to recover from. I first had it a couple of weeks ago, but I have no idea what the trigger is yet. My current plan of attach is to restore an EEPROM backup via the bootloader the next time it happens to pin it down to that (will save it first if it's not when out at the field). 

https://streamable.com/6s6mw0

Report: https://discord.com/channels/839849772864503828/839849772864503832/890142933993529364

0crap, 2021-10-04:
Subscribed.

pfeerick, 2021-11-11:
@trnnwnnhe Your "Infinite boot after removing battery" issue is most likely related to this... ;) And FYI, PR is for code contribution, not questions/reporting problems ;)

raphaelcoeffic, 2022-10-16:
Is that actually still applicable? I haven't heard any reports of something similar in a great while.

JyeSmith, 2022-10-16:
I actually flashed a recent nightly and it came back to life!  I'll flash the RC tonight and confirm, then this issue could be closed.

pfeerick, 2022-10-17:
Haven't heard/or experienced this in a while either (although admittedly haven't used those transmitters anywhere near as much of my others since around 2.6 onwards). So hopefully it is gone now - not that people stopped ETX with those transmitters 😮

closed 2022-10-17 by raphaelcoeffic

JyeSmith, 2022-10-17:
Tried the Flasher again and could not get the x7 to boot with 2.7 and 2.6.  Then flashed 2.7 with buddy and it worked.  Flashing nightly 0c4f0083 from the SD card also works.

Might be a Flasher issue 😐

pfeerick, 2022-10-17:
With Flasher, are you doing DFU or SD -> bootloader flashing? As I would _hope_ Flasher behaves itself for DFU flashing 😮.

JyeSmith, 2022-10-17:
Always dfu.

pfeerick, 2022-10-17:
Only other thing I can think of off-hand is there was an issue with x7 and x7-ACCESS firmware selection... but IIRC that was a bug in Flasher fixed probably this time last year, so it's well and truly in the bad books if it's scrambling stuff with DFU updates... 😭

JyeSmith, 2022-10-17:
I'll download the Flasher again for another try.  Cant remember the last time I updated it.

JyeSmith, 2022-10-17:
Downloading a recent Flasher fixed it 👍

pfeerick, 2022-10-17:
Thank goodness... and thank you **_very_** much for testing! :) That is one nice thing with ELRS Configurator... it told you there was a newer version and now offers to download and auto-update...